MarginallyClever/Makelangelo-software #264: Makelangelo Software not supported in current version of Catalina

toyoxx, 2020-01-22:
Good afternoon.

The .jar version of Makelangelo software cannot be run in current Catalina versions, apparently because of UI updates being performed outside of the Main Thread. It all seems to point that since the introduction of the [Main Thread Checker](https://developer.apple.com/documentation/code_diagnostics/main_thread_checker), all calls done outside of this thread kills the program. The error is as follows: 

`2020-01-22 16:13:20.053 java[31505:1390263] Apple AWT Internal Exception: NSWindow drag regions should only be invalidated on the Main Thread!
2020-01-22 16:13:20.054 java[31505:1390263] *** Terminating app due to uncaught exception 'NSInternalInconsistencyException', reason: 'NSWindow drag regions should only be invalidated on the Main Thread!'
*** First throw call stack:
(
	0   CoreFoundation                      0x00007fff351058ab __exceptionPreprocess + 250
	1   libobjc.A.dylib                     0x00007fff6b376805 objc_exception_throw + 48
	2   CoreFoundation                      0x00007fff3512e30c -[NSException raise] + 9
	3   AppKit                              0x00007fff3227e538 -[NSWindow(NSWindow_Theme) _postWindowNeedsToResetDragMarginsUnlessPostingDisabled] + 310
	4   AppKit                              0x00007fff32265ed5 -[NSWindow _initContent:styleMask:backing:defer:contentView:] + 1416
	5   AppKit                              0x00007fff32265947 -[NSWindow initWithContentRect:styleMask:backing:defer:] + 42
	6   libnativewindow_macosx.jnilib       0x000000012bf303fe Java_jogamp_nativewindow_macosx_OSXUtil_CreateNSWindow0 + 398
	7   ???                                 0x000000010f9b0c90 0x0 + 4556786832
)
libc++abi.dylib: terminating with uncaught exception of type NSException
Abort trap: 6
`
My guess is that all UI calls in macOS will have to be done using a `DispatchQueue.main.async {}` block?

Best regards,

i-make-robots, 2020-01-31:
I've never heard of  a DispatchQueue.main.async {} block.  I will look into this more.  I don't have any devices new enough to run Catalina :(

i-make-robots, 2020-02-06:
What about using https://github.com/sgothel/jogl/blob/master/src/newt/classes/com/jogamp/newt/util/MainThread.java ?

sgothel, 2020-02-06:
I also have done my duty to validate against
the MainThreadChecker, which can be enabled via property set
'-Dnativewindow.debug.OSXUtil.MainThreadChecker'

That said, which JOGL version do you use?

Please test w/ <https://jogamp.org/deployment/v2.4.0-rc-20200202/>
which I have posted here
<http://forum.jogamp.org/JogAmp-Jbeil-RC-Build-2-4-0-rc-20200202-td4040323.html>

Cheers, Sven

i-make-robots, 2020-02-06:
according to the pom we have two settings:

	<dependency>
	  <groupId>org.jogamp.fat</groupId>
	  <artifactId>jogamp-fat</artifactId>
	  <version>2.4.0</version>
	</dependency>

and 

  <properties>
    <jogl.version>2.3.2</jogl.version>
  </properties>

I have changed the second to 2.4.0 and now I need to coordinate with @toyoxx to test the new build.

i-make-robots, 2020-02-06:
@toyoxx please DM me and I'll send you a new build that has the proposed change.

sgothel, 2020-02-06:
I wonder who published our jogamp to which maven server?

sgothel, 2020-02-06:
> I wonder who published our jogamp to which maven server?

also there is no jogamp 2.4.0 released yet, just the RC

toyoxx, 2020-02-07:
@i-make-robots Cool! I'll DM you so I can try on my PC!

i-make-robots, 2020-02-07:
@toyoxx please mail dan@marginallyclever.com.  I'll send you a dropbox link.

@sgothel having jogamp in maven is THE BEST.  I love it.  I would gladly match against the -rc version.  Could publications be automated by your build system?

i-make-robots, 2020-06-12:
glClear() is the method that is supposed to wipe the old image so the new one can be drawn.  Is it not being called?  Is the COLOR_BUFFER_BIT not working?  Hm...

i-make-robots, 2020-07-03:
Is this issue ongoing?

Jip-Hop, 2020-07-05:
Makelangelo opens fine on Catalina. It's just this drawing glitch. Not a huge deal since it only happens when zooming out a lot. The stuff that matters (within machine bounds) is clearly visible. Tried with the latest nightlies. By the way, why do the nightlies have "windows" in their name: "makelangelo_windows-nightly-20200703-f8284b.jar"? It's cross platform right?

i-make-robots, 2020-07-05:
It’s compiled on windows and might not include the native code for the other platforms. If you say it does and it runs on the others ok then I’ll change the name going forward.

Jip-Hop, 2020-07-05:
Aren't all releases compiled on Windows then? I can confirm the nightly builds run just fine on MacOS.

i-make-robots, 2020-07-05:
K, I’ll update the action so future names don’t have “windows” in them.