MarginallyClever/Makelangelo-software #267: Export .jar File in Eclipse

bodely, 2020-02-19:
Hey,

I tried different types of exporting the Makelangelo software in Eclipse as a .jar file but everytime when I want to start the .jar file, only a window opens that says 'application is not runnable'.

Could you please help me and tell me the correct procedure of exporting the software to a .jar file that works?

Thank you very much! Felix

i-make-robots, 2020-02-19:
Los the project in eclipse by importing a maven project. Then build it with target “package”.

bodely, 2020-02-19:
Thanks for your answer. I can compile and run the software in Eclipse but when I want to run the exported .jar file I get some errors (see the photos). Do you know where the problem is?
Thanks Felix :)
[image]
[image]

i-make-robots, 2020-02-19:
Huh.  Probably translator is failing to find the resource file that
includes all the text that appears in the app.
I misspoke before.  make sure the Maven Build > Main > Goals: "package".

That should correctly put everything into the jar-with-dependencies and
then the translator *should* work.
Please keep me updated!  I want to make this easier for everyone.

bodely, 2020-02-20:
Hey Dan, thanks a lot now it works perfectly. I am just a beginner with maven, I built the project the wrong way before you gave me the hint. 
Felix :)

i-make-robots, 2020-02-20:
Where should we put the hint so everyone finds it?

What are your next goals? I’m here to help.

bodely, 2020-02-21:
I think a tutorial how to build and export the software in the Readme file of the software is helpful?

I would really like to control the Makelangelo via Wifi. Controling it with any device (even smartphphone) without the .jar software would be very nice but I don't have that much time at the moment to realize this...

Thanks for your help Dan :)

i-make-robots, 2020-02-21:
TCP/IP connections exist in the software. We have yet to write the firmware for, say, esp32.

i-make-robots, 2020-02-24:
would "If you would like to compile the code yourself, please see https://github.com/MarginallyClever/Makelangelo/wiki/Home/" in the README work for you? we have steps on the wiki already.

i-make-robots, 2020-07-03:
@bodely we now have nightly builds so that you don't have to build it yourself to try the new hotness.  
Having said that, I encourage you to improve the code with your ideas and send in pull requests.
Cheers!

closed 2020-07-03 by i-make-robots