MarginallyClever/Makelangelo-software #268: Makelangelo shows boot screen then closes [Mac OS Mojave]

Ohmarinus, 2020-05-04:
Hi,

So after upgrading my OS, I'm back to working with Makelangelo. It worked really well in the past but now that I'm using Mojave it starts up, shows the boot screen with the Makelangelo plotter and then quits 'unexpectedly'.

Now I'm not a developer but I found this to be the culprit:
`Crashed Thread:        32  Java: AWT-EventQueue-0

Exception Type:        EXC_BAD_ACCESS (SIGABRT)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000008
Exception Note:        EXC_CORPSE_NOTIFY`

Any idea if there is a fix for this?
Using Java 1.8.0_252, x86_64

i-make-robots, 2020-05-04:
I’ve never seen this error before. Can you run the dev branch from eclipse and find where it happens?  Did google suggest anything about the error?

Ohmarinus, 2020-05-05:
I'll try to run the DEV branch, but I'm not familiar with building a JAR so I'll have to figure that out first :)

i-make-robots, 2020-05-05:
The wiki on github has a guide.

Jip-Hop, 2020-06-05:
I'm using Java Version 8 Update 251 (build 1.8.0_251-b08) on Mojave 10.14.6 and have no issues running Makelangelo-7.21.0-with-dependencies.jar. I think I downloaded and installed Java from [java.com](https://www.java.com/en/download/mac_download.jsp).

i-make-robots, 2020-06-05:
What does the log.txt say?

i-make-robots, 2020-06-25:
are #268 and #271 the same thing?

i-make-robots, 2020-07-03:
@Ohmarinus can you please try https://github.com/MarginallyClever/Makelangelo-software/releases/download/Nightly/makelangelo_windows-nightly-20200703-f8284b.jar , our testing build?

Ohmarinus, 2020-07-04:
Hi Dan @i-make-robots I can confirm this works yay!

Log.txt:
`2020-07-04 20:03:42 PROGRAM START
2020-07-04 20:03:42 OS=mac os x
2020-07-04 20:03:42 Locale=en_NL
2020-07-04 20:03:42 starting preferences...
2020-07-04 20:03:42 makelangelo.version=7.23.0
2020-07-04 20:03:42 starting robot...
2020-07-04 20:03:42 starting transfer handler...
2020-07-04 20:03:42 starting connection manager...
2020-07-04 20:03:42 starting translator...
2020-07-04 20:03:42 Default language = English
2020-07-04 20:03:42 Looking for translations in jar:file:/Users/marinusdebeer/Downloads/makelangelo_windows-nightly-20200703-f8284b.jar!/languages
2020-07-04 20:03:42 rootDir=/Users/marinusdebeer/Downloads
2020-07-04 20:03:42 Found Jar:languages/piglatin.xml
2020-07-04 20:03:42 Found Jar:languages/english.xml
2020-07-04 20:03:42 Found Jar:languages/dutch.xml
2020-07-04 20:03:42 Creating GUI...
2020-07-04 20:03:42 Create menu bar
2020-07-04 20:03:42   file...
2020-07-04 20:03:42   view...
2020-07-04 20:03:42   help...
2020-07-04 20:03:43   finish...
2020-07-04 20:03:43   adding menu bar...
2020-07-04 20:03:43 create content pane...
2020-07-04 20:03:43   get GL capabilities...
2020-07-04 20:03:44   create draw panel...
2020-07-04 20:03:44   set robot...
2020-07-04 20:03:44   assign panel to robot...
2020-07-04 20:03:45 Cannot run raspistill
2020-07-04 20:03:45   create log panel...
2020-07-04 20:03:45   vertical split...
2020-07-04 20:03:45   horizontal split...
2020-07-04 20:03:45   tweak...
2020-07-04 20:03:45 adjust window size...
2020-07-04 20:03:45   make visible...
2020-07-04 20:03:45   adding drag & drop support...
2020-07-04 20:03:45   starting animator...
2020-07-04 20:03:45 checking sharing permissions...
`

i-make-robots, 2020-07-04:
Woo!  Nightlies paying off already!

closed 2020-07-04 by i-make-robots