MarginallyClever/Makelangelo-software #271: Closing on Windows 10

Jip-Hop, 2020-06-07:
Hi Dan,

I've exported a JAR file from the dev branch according to the [Exporting a jar file for Release instructions](https://github.com/MarginallyClever/Makelangelo-software/wiki/Exporting-a-jar-file-for-Release).

It builds the JAR, but with [some errors](https://pastebin.com/eD8UA8aG) (Javadoc).

The exported JAR works fine on MacOS, but closes after the splash screen on Windows 10. Here's the log.txt output when running on Windows:

```
2020-06-07 11:16:52 PROGRAM START
2020-06-07 11:16:52 OS=windows 10
2020-06-07 11:16:52 Locale=en_US
2020-06-07 11:16:52 starting preferences...
2020-06-07 11:16:52 makelangelo.version=7.23.0
2020-06-07 11:16:52 starting robot...
2020-06-07 11:16:52 starting transfer handler...
2020-06-07 11:16:52 starting connection manager...
2020-06-07 11:16:52 starting translator...
2020-06-07 11:16:52 Looking for translations in jar:file:/D:/User/Downloads/Makelangelo-7.23.0-with-dependencies.jar!/languages
2020-06-07 11:16:52 rootDir=D:\User\Downloads
2020-06-07 11:16:52 Found Jar:languages/piglatin.xml
2020-06-07 11:16:53 Found Jar:languages/dutch.xml
2020-06-07 11:16:53 Found Jar:languages/english.xml
2020-06-07 11:16:53 starting GUI...
```

I've noticed the same behaviour when trying to run Makelangelo 7.22.6 experimental from the [Makelangelo dev blog 2020 #2](https://www.marginallyclever.com/2020/01/makelangelo-dev-blog-2020-2/).

Any clue?

i-make-robots, 2020-06-07:
Well... I added more log files right between "starting GUI" and "checking sharing permissions..." to see what is the last line run before the silent crash.  

My export build completes without javadoc errors so I can't replicate that bug.

Jip-Hop, 2020-06-07:
New log output:

```
2020-06-07 21:15:01 PROGRAM START
2020-06-07 21:15:01 OS=windows 10
2020-06-07 21:15:01 Locale=en_US
2020-06-07 21:15:01 starting preferences...
2020-06-07 21:15:01 makelangelo.version=7.23.0
2020-06-07 21:15:01 starting robot...
2020-06-07 21:15:01 starting transfer handler...
2020-06-07 21:15:01 starting connection manager...
2020-06-07 21:15:01 starting translator...
2020-06-07 21:15:01 Default language = English
2020-06-07 21:15:01 Looking for translations in jar:file:/D:/User/Downloads/Makelangelo-7.23.0-with-dependencies.jar!/languages
2020-06-07 21:15:01 rootDir=D:\User\Downloads
2020-06-07 21:15:01 Found Jar:languages/piglatin.xml
2020-06-07 21:15:01 Found Jar:languages/english.xml
2020-06-07 21:15:01 Found Jar:languages/dutch.xml
2020-06-07 21:15:01 Creating GUI...
2020-06-07 21:15:01 Create menu bar
2020-06-07 21:15:01   file...
2020-06-07 21:15:01   view...
2020-06-07 21:15:01   help...
2020-06-07 21:15:01   finish...
2020-06-07 21:15:01   adding menu bar...
2020-06-07 21:15:01 create content pane...
2020-06-07 21:15:01   get GL capabilities...
```

i-make-robots, 2020-06-07:
Really!  So it fails on the first call to Jogamp.

Jip-Hop, 2020-06-08:
In Eclipse on MacOS I get a warning related to Jogamp. But here it doesn't silently crash and instead continues to work.
```
log dir=/Foo/bar/Makelangelo-software/
2020-06-08 18:13:28 PROGRAM START
2020-06-08 18:13:28 OS=mac os x
2020-06-08 18:13:28 Locale=en_NL
2020-06-08 18:13:28 starting preferences...
2020-06-08 18:13:28 makelangelo.version=7.23.0
2020-06-08 18:13:28 starting robot...
2020-06-08 18:13:28 starting transfer handler...
2020-06-08 18:13:28 starting connection manager...
2020-06-08 18:13:28 starting translator...
2020-06-08 18:13:28 Default language = English
2020-06-08 18:13:28 Looking for translations in file:/Foo/bar/Makelangelo-software/target/classes/languages
2020-06-08 18:13:28 rootDir=/Foo/bar/Makelangelo-software
2020-06-08 18:13:28 Found /Foo/bar/Makelangelo-software/target/classes/languages/piglatin.xml
2020-06-08 18:13:28 Found /Foo/bar/Makelangelo-software/target/classes/languages/dutch.xml
2020-06-08 18:13:28 Found /Foo/bar/Makelangelo-software/target/classes/languages/english.xml
2020-06-08 18:13:28 Found /Foo/bar/Makelangelo-software/pom.xml
2020-06-08 18:13:28 ERROR Failed to load /Foo/bar/Makelangelo-software/pom.xml
2020-06-08 18:13:28 Creating GUI...
2020-06-08 18:13:29 Create menu bar
2020-06-08 18:13:29   file...
2020-06-08 18:13:29   view...
2020-06-08 18:13:29   help...
2020-06-08 18:13:29   finish...
2020-06-08 18:13:29   adding menu bar...
2020-06-08 18:13:29 create content pane...
2020-06-08 18:13:29   get GL capabilities...
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by com.jogamp.common.os.NativeLibrary$3 (file:/Foo/bar/.m2/repository/org/jogamp/fat/jogamp-fat/2.4.0/jogamp-fat-2.4.0.jar) to method java.lang.ClassLoader.findLibrary(java.lang.String)
WARNING: Please consider reporting this to the maintainers of com.jogamp.common.os.NativeLibrary$3
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
2020-06-08 18:13:30   create draw panel...
2020-06-08 18:13:30   set robot...
2020-06-08 18:13:30   assign panel to robot...
2020-06-08 18:13:30 Cannot run raspistill
2020-06-08 18:13:30   create log panel...
2020-06-08 18:13:31   vertical split...
2020-06-08 18:13:31   horizontal split...
2020-06-08 18:13:31   tweak...
2020-06-08 18:13:31 adjust window size...
2020-06-08 18:13:31   make visible...
2020-06-08 18:13:31   adding drag & drop support...
2020-06-08 18:13:31   starting animator...
2020-06-08 18:13:31 checking sharing permissions...
```

i-make-robots, 2020-06-08:
here's someone who had a janky fix.  https://stackoverflow.com/questions/10995894/worldwind-runnable-jar-no-gluegen-rt-in-java-library-path

Jip-Hop, 2020-06-08:
I don't like this issue :stuck_out_tongue:
What platform do you run on? Can you reproduce the silent crash issue?

i-make-robots, 2020-06-08:
i'm on windows 10.  runs fine here.

Jip-Hop, 2020-06-08:
I've got another Janky fix.

Download [the latest aggregated autobuild of jogamp-fat.jar](https://jogamp.org/deployment/archive/master/gluegen_937-joal_660-jogl_1506-jocl_1147/fat/jogamp-fat.jar).

Rename it to `jogamp-fat-2.4.0.jar`.
Overwrite the `jogamp-fat-2.4.0.jar` in the `Makelangelo-software/java/local-maven-repo/org/jogamp/fat/jogamp-fat/2.4.0/` folder.

The exported `Makelangelo-7.23.0-with-dependencies.jar` now runs without silently crashing.

I did this with a freshly imported project in Eclipse, and it was the first built I ran.

i-make-robots, 2020-06-08:
yeah... the jogbamp team has said the fat is janky and (apparently) has a version of jogamp not released yet.  super dodgy.

i-make-robots, 2020-06-08:
to be clear... I used to have the fat jar in the project until they were like "woah, wtf is this..."

Jip-Hop, 2020-06-08:
Maybe go back to that approach, hehe. I'm a bit sad I can't really compile Makelangelo and run it on my Windows machine. 🙁 

Strange though that it runs for you, on Windows 10, whereas it doesn't run for me. I've got Windows 10 Pro 1909 OS build 18363.836.

Your Makelangelo 7.22.6 build from January silently crashes for me too, so it's already in there for a little while.

Maybe different Java versions? I've got Version 8 Update 251 (build 1.8.0_251-b08).

i-make-robots, 2020-06-08:
I can run it from both.  I'm using JavaSE-1.8 (jdk1.8.0_65)

Jip-Hop, 2020-06-08:
I just installed JDK 8u251 and now I can run your Makelangelo 7.22.6 build from January without it silently crashing.

Jip-Hop, 2020-06-08:
I uninstalled all Java versions and now tried with `jdk-14.0.1_windows_x64_bin.exe`, and that works too. So I guess the problem was that I hadn't installed a JDK just JRE. 😄

i-make-robots, 2020-06-08:
were you running the app from within eclipse?  without a jdk there's no
hooks for the debugger.

i-make-robots, 2020-06-08:
Please add a note to the wiki page for developers so they can avoid the same issue.

Jip-Hop, 2020-06-08:
I was building in Eclipse on macos and running the exported jar on Windows. It wouldn't work on Windows without the jdk installation (however, the latest official Makelangelo release works with just jre and no jdk, that was confusing). On Windows I haven't installed Eclipse.

Perhaps the next release needs JDK for people who want to run Makelangelo?

i-make-robots, 2020-07-03:
Is this issue closed now?

i-make-robots, 2020-07-03:
Duplicate of #268

closed 2020-07-03 by i-make-robots