MarginallyClever/Makelangelo-software #273: Home position drifting after settings change

Jip-Hop, 2020-06-08:
After changing settings, with the pen holder in any position other than home, the current position will become home position. As a result, the pen will not draw in the correct place. It will miss the paper and may crash into the ceiling when moving up. It tries to go higher than is actually possible (both belts are pulled fully tight).

Steps to reproduce:
- Connect to the machine
- Click find home
- Manually drive the pen holder up, e.g. 20cm
- Click Settings, change e.g. Draw Speed under Pen tab
- Click OK
- Makelangelo GUI now shows the pen holder in the home position, but in reality the pen holder is 20cm above home

Alternative steps:
- Connect to the machine
- Click find home
- Manually drive the pen holder up, e.g. 20cm
- Makelangelo GUI now shows the pen holder in the home position, but in reality the pen holder is 20cm above home

Possible solutions:

Makelangelo GUI should not reset position after changing a setting. Or if the firmware keeps track of its current position, then Makelangelo GUI should query that position and use that instead of assuming the machine is in the home position. Or if that's not possible, run find-home for machines with endstops, or else show a warning to the user to manually put the machine in its home position.

Before settings change:
[image]

After settings change:
[image]

i-make-robots, 2020-06-08:
IMO the firmware saves settings way too often, even when nothing has
changed.  The EEPROM (tiny hard drive) only has ~100k writes before it
becomes unwritable, so minimizing this would be good.  I think the firmware
resets the position to default whenever the limits are changed - by that i
mean whenever the "change limits" gcode is called, even if the limit values
don't change.

Jip-Hop, 2020-06-08:
Not saving if nothing changed sounds good :)

What is the "change limits" gcode? Can this be fixed with a check if the new limits are the same as the current limits, and if they're the same not calling the "change limits" gcode?

Jip-Hop, 2020-06-08:
If the limits do change though, I think it would be nice to run 'find home' on machines with endstops. Or is that a bad idea?

And for other machines show a warning to put the pen holder manually in the home position.

i-make-robots, 2020-06-08:
yeah... some kind of "unhomed" state that forces everyone to rehome.

Jip-Hop, 2020-06-11:
Hi Dan,

> IMO the firmware saves settings way too often, even when nothing has changed.

Has this already been solved with [commit f9835b8](https://github.com/MarginallyClever/Makelangelo-firmware/commit/f9835b8) in the dev branch of the Makelangelo-firmware?

i-make-robots, 2020-06-11:
Maybe a little?  There are other EEPROM_write* methods.  also being at the
very last stage there are probably messages saying "writing value" when
nothing happens (because the number didn't change).  is a bit misleading...
but I guess if it ain't broke don't fix it...

i-make-robots, 2020-06-11:
Maybe this will help us talk about stuff better: https://discord.gg/t57JZG

i-make-robots, 2020-07-05:
So looking into this a bit more.... I think it would be safe to assume that whatever XYZ position the plotter has before a config change should stay the same UNLESS it somehow becomes out of bounds.  So... when the edge case happens, what do?
a) bound old position to new limits?
b) set position to "invalid, must re-home"?

Jip-Hop, 2020-07-05:
I agree that would be an edge case. I haven't changed my machine dimensions at all since I got it setup properly. But I do change paper size and speed settings, so I'm glad you're looking into it. Would be great if I wouldn't have to re-home after those changes :)

I'd go for option b, but maybe it could be prevented in some cases too (if by mistake or user is unaware of consequence). Before applying the new bounds, it could give a choice:

"Current pen position lies outside of new machine dimensions. Re-homing will be necessary. Are you sure you wish to apply the new machine dimensions?" With choices: "Cancel" or "Yes".

And for machines that can auto find home, after choosing "Yes", it could ask: "Should the machine auto-home right now?"

By the way, Makelangelo will also forget its current position when disconnecting and reconnecting from the GUI. The machine is powered on all the time, so it feels like it should be able to remember and report its position when connecting again.

i-make-robots, 2020-07-05:
Choices "Yes" or "No".  "Cancel" is ambiguous?

What would the unit test look like? try { (new Makelangelo5()).teleport(edge).getSettings().changeDimensions(edge/2) } catch() ?

Jip-Hop, 2020-07-05:
I don't know if there need to be 3 choices, it that's what you mean. "Cancel" or "No" would do the same I guess in this case.

i-make-robots, 2020-07-05:
I mean “no” is better than “cancel”. Cancel what? You can cancel making a choice. Be bold, pick an option :)

Jip-Hop, 2020-07-05:
Yes and No it is then!

closed 2021-10-15 by i-make-robots