MarginallyClever/Makelangelo-software #278: Calibrating a Makelangelo 3.3 like machine

Jip-Hop, 2020-06-12:
Hi Dan,

I've made [these changes](https://github.com/Jip-Hop/Makelangelo-firmware/commit/aaeefd633f513a9703ffe2b6fb1627fd33d1a5c2) in the firmware for my machine, which uses Arduino Mega + Ramps 1.4 + DRV8825, has endstops but no LCD or SD.

It seems to work properly but I can't get the calibration just right. I've already [fixed the issue](https://github.com/MarginallyClever/Makelangelo-software/pull/276) where the software would revert to version 2, instead of 3.3, which was also [reported on the forum](https://www.marginallyclever.com/learn/forum/topic/custom-build-with-ramps-and-auto-homing/#post-27587).

Can you instruct me how to calibrate a Makelangelo 3.3 like machine of custom size? Are the firmware changes enough, or do I need to change the values in [makelangelo33Setup()](https://github.com/MarginallyClever/Makelangelo-firmware/blob/e6b92e482b9e83ff75539cab25ba94a2513d8c6d/Makelangelo-firmware.ino#L997) as well?

Should I manually move the pen holder to a specific place for the 3.3 calibration process?

Which gcode commands, with which measurements, should I run after connecting with the software? And in which order?

Thanks in advance 🙂

Jip-Hop, 2020-06-12:
How I've tried to calibrate is as follows:

- Pull the pen holder down till both endstops trigger
- Measure the distance from the center of the motors to the center of the pen holder (226cm on both sides)
- Send `D7 L2260 R2260`
- Send `D9`

Results are pretty good, except for the following:

I put the pen holder in the center, at the bottom of the paper. Then with manual driving I press 10 times the +100 button for the Y-axis. I measure the travelled distance. I expect the travelled distance to be 100cm, but it actually moved up about 93cm.

Only the Y-axis is off... Doing the same for the X-axis gives correct results.

i-make-robots, 2020-06-12:
As a rule of thumb it's probably better to put your custom configuration on
a different branch and then update that branch when needed.
this way the main branch stays on default settings.

The measurement would be best if the belt turned over a point (circle with
radius 0).  Unfortunately it rolls over a circle, which changes the angle
as the belt leaves the pulley.  In some cases the belt then goes around one
of the stand offs for the motor, meaning it is more predictable until the
pen holder is below the point where the belt touches the stand off.
These things all contribute to wrecking the kinematic model.

For that reason motor center to center is probably not the best
measurement.  I suspect inside bottom corners (the stand offs) will get
better results.
fwiw, @buildlog worked on a version where the belt went over a high point
of inflection so the kinematics were better.

Jip-Hop, 2020-06-12:
Hi Dan, thanks as always :)

My firmware changes are in their own branch named [jip](https://github.com/Jip-Hop/Makelangelo-firmware/tree/jip).

I've mounted my motors from the top, without stand offs. The belt leaves the pulley without any obstruction. So in my case that's not what's causing the 7cm difference.

[image]
[image]

Can I do anything to tune my machine better? Measure from different points? Try a different command? Or is this as good as it gets with the current implementation?

I'm trying to calibrate it right, because I've bought a 1x2m whiteboard sticker. I can only stick it on the wall once. 😝 So I hope to get it in the right place.

i-make-robots, 2020-06-12:
Well you definitely know your machine width, so my next suspicion is the belt length.  we cut 1m belts for the m5 and yet I have to set a belt length of 1025mm.

i-make-robots, 2020-06-12:
Having said that... I'm very open to any ideas you have that would produce more accurate (square) drawings.

Jip-Hop, 2020-06-15:
Good to know. I was able to tweak the values for `D7` through trial and error and the results are more accurate now.

closed 2020-06-15 by Jip-Hop

i-make-robots, 2020-06-15:
Any thoughts about how to calculate this to get it more accurate every time?

Jip-Hop, 2020-06-15:
Not really, I'm no calculation wizard to be honest.

But I guess it could be possible with a calibration procedure. Mark two points, point A and B, on the X axis with a known distance. Say 1 meter from A to B. Manually drive the pen holder to point A. Click a button in the GUI to mark point A as start position. Manually drive to point B (the firmware or desktop software remember the number of 'units' traveled from A to B). Click a button in the GUI to stop measures once at point B.

Repeat process for Y axis.

Now comes the magic math I've no clue about, which works out the internal calibration based on the data gathered.

My manual proces was similar. I globally measured the values for D7. Then checked if it traveled 1 meter in X and Y direction. Increase D7 values if it was traveling less then 1 meter, until I found the rights values.

i-make-robots, 2020-06-15:
Yeah, that makes sense.  One of the reasons I stopped making the 3 was that
every user had to learn how to calibrate it.  Most people just want to get
on with plotting.