MarginallyClever/Makelangelo-software #280: Detailed SVG import error

Jip-Hop, 2020-06-21:
When importing an SVG file with small details, it's imported incorrectly. Lines are drawn between points, where there shouldn't be a line.

[image]

The test file is made in Adobe Illustrator, exported as SVG. The SVG imports with the error just mentioned. However exporting the same design from Adobe Illustrator as DXF, and importing that into Makelangelo, works just fine. So the error is with SVG only.

[image]

The problem can't be the SVG file itself. I opened the SVG in [LightBurn](https://lightburnsoftware.com), and exported from there as gcode. I manually inserted a pendown command and changed the extension from `.gc` to `.ngc`. Importing this gcode into Makelangelo is now free of errors. Since this gcode is generated, without errors, by LightBurn from the SVG file which was causing issues in Makelangelo, the issue can't be with the SVG file itself.

[image]

I'm using the latest release, Makelangelo-7.23.0-with-dependencies.jar, for these tests. All screenshots are made fully zoomed in.

[SVG DXF NGC test files.zip](https://github.com/MarginallyClever/Makelangelo-software/files/4809256/SVG.DXF.NGC.test.files.zip)

Jip-Hop, 2020-06-22:
[Commit 48988d0](https://github.com/MarginallyClever/Makelangelo-software/commit/48988d03a36330ea26124489bec790e46b3bf92b) hasn't fixed this error for me yet. Lines are still drawn where they shouldn't on SVG import.

i-make-robots, 2020-06-24:
Opened the test SVG.  Looks nothing like the screenshots above.  Please resend.

i-make-robots, 2020-06-24:
I have a fix.  It will mean many more pen-up moves on very small, adjacent segments.  Ideally an optimization step comes after the import and sorts all that crud out.

i-make-robots, 2020-06-24:
> Hahah yes sorry it's tiny! Don't know at what size it starts to happen but
> I noticed it in a section of a large drawing.

Yeah.  an optimization I was trying was "don't lift the pen on very short
travels".  You're technically right that it shouldn't be done at load time,
it should be a late-game process of some kind.
If the bug is fixed, please close the issue(s).

Jip-Hop, 2020-06-25:
It's fixed now :)

It does sound like a valuable optimization. Basically it joins paths based on a distance tolerance. Could that perhaps become a checkbox in the Art Pipeline, like Reorder and Simplify?

But the "don't lift the pen on very short travels" optimization would only work properly in combination with path reordering, else this error occurs again (additional lines are drawn between points where there shouldn't be any line).

closed 2020-06-25 by Jip-Hop