MarginallyClever/Makelangelo-software #281: DXF export upside down

Jip-Hop, 2020-06-21:
Fibonacci Spiral Fractal generated with latest release of Makelangelo: Makelangelo-7.23.0-with-dependencies.jar:

[image]

Upside down when exported as DXF and imported again:

[image]

DXF import the correct way up, so the issue has to be with the export function.

i-make-robots, 2020-06-22:
Is this better now, @Jip-Hop ?

Jip-Hop, 2020-06-22:
Yes, you fixed it!

closed 2020-06-22 by Jip-Hop