MarginallyClever/Makelangelo-software #282: Some shapes/lines in SVGs won't import

Jip-Hop, 2020-06-21:
What the SVG file looks like in Illustrator:
[image]

What is imported in Makelangelo:
[image]

It's missing the straight line, square and circle.

I don't think it has anything to do with Illustrator. I saved [this example SVG code](https://developer.mozilla.org/en-US/docs/Web/SVG/Element/rect) as SVG file and loaded it in Makelangelo. It didn't import anything.

Would be great to get import/export issues fixed, so it becomes easier to round trip and combine stuff generated in Makelangelo with things drawn in other programs.

How can I help?

i-make-robots, 2020-06-21:
loadAndSaveSVG.java is where the action happens, in parseAll().
At the moment I believe only types supported are path, polyline, and
polygon.  It would be better to gram all drawable elements, convert them to
path, and proceed with existing code.

My question is: how would we unit test this to make sure it keeps working?

i-make-robots, 2020-06-22:
[image]
Latest dev commit should help.  Can you confirm?

i-make-robots, 2020-06-22:
src/test/resources/test.svg will have this image.

Jip-Hop, 2020-06-22:
Hi Dan, thanks for looking into it so quickly.

I'm able to load your test svg file.
But it didn't fix the issue for any of my test svgs.

I get this error constantly:

```
2020-06-22 10:38:06 Loading...
2020-06-22 10:38:06 Failed to load some elements (1)
```

For example the `import_error_test.svg` file from issue #280 now throws above error and doesn't import any lines.

I forgot to attach my shapes test svg to my initial post. I've included it in [this zip file](https://github.com/MarginallyClever/Makelangelo-software/files/4812186/more.test.svgs.zip), along with another rectangle test and circle test. None of them import any lines for me.

By the way I had to turn on `Skip tests` in Eclipse, else it wouldn't compile.

```
[INFO] 
[INFO] Results:
[INFO] 
[ERROR] Errors: 
[ERROR]   SelectTests.testFloat:177 » NumberFormat For input string: "0,1"
[INFO] 
[ERROR] Tests run: 9, Failures: 0, Errors: 1, Skipped: 0
```

i-make-robots, 2020-06-22:
Your machine is set to what country?  Looks like german punctuation.

Jip-Hop, 2020-06-22:
I'm from the Netherlands, next to Germany. But I think my system is set to English. Will check later.

i-make-robots, 2020-06-22:
The test that is failing checks for german-formatted numbers.

Jip-Hop, 2020-06-22:
Aah okay. We use dots as thousand separator 1.000 and comma for decimals, like so: 10,5.

i-make-robots, 2020-06-22:
Updated tests - some have a single unit of each type, some have more than one, some are combined.
All tests pass right now.  No support for rounded corners on rectangles yet.

Jip-Hop, 2020-06-22:
Nice, I confirm circle and rectangle now work.

But a simple straight line still won't import.

```svg
<svg
  xmlns="http://www.w3.org/2000/svg"
  viewBox="0 0 163.3 47.2"
  xmlns:v="https://vecta.io/nano"
>
  <path
    d="M163.1.5L.1 46.7"
    fill="none"
    stroke="#e62227"
    stroke-miterlimit="10"
  />
</svg>
```
Importing above code won't draw anything. It's the same straight line that's missing in the image in my original post.

i-make-robots, 2020-06-22:
0375da99 might fix this.  Added your single path as a test and confirmed it looks right.

Jip-Hop, 2020-06-23:
Thanks Dan. The single line now imports correctly! But the same line still doesn't import in the document with multiple objects.

Here's the file again where the line is missing after import: [shapes_test.svg.zip](https://github.com/MarginallyClever/Makelangelo-software/files/4817835/shapes_test.svg.zip).

About rectangles with rounded corners. When exporting the SVG with rounded corners from Illustrator as DXF, and importing it in Makelangelo, the rounded corners are preserved. Would it be an idea to convert all SVGs to DXF 'under the hood' when importing in Makelangelo? Hopefully there's a rock solid Java library for this already...

Maybe [drawj2d](https://drawj2d.sourceforge.io)? It imports and exports from/to SVG and DXF. More info here: [https://drawj2d.sourceforge.io](https://drawj2d.sourceforge.io).

i-make-robots, 2020-06-23:
Ah well... that's a line, not a path.  if the SVG elements were converted to path (as is done in DXF files) then everything would be PATH or POLYLINE elements and easily supported.   I've added support for <line> items in 42093e39

Jip-Hop, 2020-06-24:
Thanks for the explanation, I get it now. And thanks for making it work :) It's just #280 now that's messing up my imports. Hope you can have another stab at that too.

i-make-robots, 2020-06-24:
rounded corners for rect added in 363a2041

Jip-Hop, 2020-06-25:
Much improved SVG import capabilities, I love it!

closed 2020-06-25 by Jip-Hop