MarginallyClever/Makelangelo-software #283: Float international number format troubles

i-make-robots, 2020-06-22:
Your machine is set to what country?  Looks like german punctuation. 

> On Jun 22, 2020, at 1:46 AM, Jip-Hop <notifications@github.com> wrote:
> 
> ﻿
> Hi Dan, thanks for looking into it so quickly.
> 
> I'm able to load your test svg file.
> But it didn't fix the issue for any of my test svgs.
> 
> I get this error constantly:
> 
> 2020-06-22 10:38:06 Loading...
> 2020-06-22 10:38:06 Failed to load some elements (1)
> For example the import_error_test.svg file from issue #280 now throws above error and doesn't import any lines.
> 
> I forgot to attach my shapes test svg to my initial post. I've included it in this zip file, along with another rectangle test and circle test. None of them import any lines for me.
> 
> By the way I had to turn on Skip tests in Eclipse, else it wouldn't compile.
> 
> [INFO] 
> [INFO] Results:
> [INFO] 
> [ERROR] Errors: 
> [ERROR]   SelectTests.testFloat:177 » NumberFormat For input string: "0,1"
> [INFO] 
> [ERROR] Tests run: 9, Failures: 0, Errors: 1, Skipped: 0
> —
> You are receiving this because you commented.
> Reply to this email directly, view it on GitHub, or unsubscribe.

_Originally posted by @i-make-robots in https://github.com/MarginallyClever/Makelangelo-software/issues/282#issuecomment-647585845_

i-make-robots, 2020-06-22:
@Jip-Hop this is a separate issue.

i-make-robots, 2020-06-22:
@Jip-hop at the top of your log file what does "Locale" say?  it should be right after "PROGRAM START" and "OS=..."

Jip-Hop, 2020-06-22:
Here's the log file. My locale is en_NL.
Feels great to work out these issues together :)

```
2020-06-22 19:43:31 PROGRAM START
2020-06-22 19:43:31 OS=mac os x
2020-06-22 19:43:31 Locale=en_NL
2020-06-22 19:43:31 starting preferences...
2020-06-22 19:43:31 makelangelo.version=7.23.0
2020-06-22 19:43:31 starting robot...
2020-06-22 19:43:31 starting transfer handler...
2020-06-22 19:43:31 starting connection manager...
2020-06-22 19:43:31 starting translator...
2020-06-22 19:43:31 Default language = English
2020-06-22 19:43:31 Looking for translations in jar:file:/GitHub/Makelangelo-software/target/Makelangelo-7.23.0-with-dependencies.jar!/languages
2020-06-22 19:43:31 rootDir=/GitHub/Makelangelo-software/target
2020-06-22 19:43:31 Found Jar:languages/piglatin.xml
2020-06-22 19:43:31 Found Jar:languages/dutch.xml
2020-06-22 19:43:31 Found Jar:languages/english.xml
2020-06-22 19:43:32 Creating GUI...
2020-06-22 19:43:32 Create menu bar
2020-06-22 19:43:32   file...
2020-06-22 19:43:32   view...
2020-06-22 19:43:32   help...
2020-06-22 19:43:32   finish...
2020-06-22 19:43:32   adding menu bar...
2020-06-22 19:43:32 create content pane...
2020-06-22 19:43:32   get GL capabilities...
2020-06-22 19:43:33   create draw panel...
2020-06-22 19:43:33   set robot...
2020-06-22 19:43:33   assign panel to robot...
```

i-make-robots, 2020-06-22:
I'm writing a better test now that confirms the number format system works across all locales.

i-make-robots, 2020-06-22:
bdd34089 improved SelectFloat and tests should pass.

Jip-Hop, 2020-06-22:
> bdd3408 improved SelectFloat and tests should pass.

Latest dev branch now compiles for me with all tests passed.

i-make-robots, 2020-06-22:
Woohoo!

closed 2020-06-22 by i-make-robots