MarginallyClever/Makelangelo-software #284: Crash at start up Windows 10

bjbsquared, 2020-06-24:
Shows MakelAngelo image and then stops.

running start windows.bat at the Exception:

2020-06-24 13:55:12 Found Jar:languages/dutch.xml
2020-06-24 13:55:12 Found Jar:languages/english.xml
2020-06-24 13:55:12 Found Jar:languages/piglatin.xml
2020-06-24 13:55:12 starting GUI...
Exception in thread "AWT-EventQueue-0" java.lang.UnsatisfiedLinkError: jogamp.common.jvm.JVMUtil.initialize(Ljava/nio/ByteBuffer;)Z
        at jogamp.common.jvm.JVMUtil.initialize(Native Method)

Jip-Hop, 2020-06-25:
I think this is a duplicate of #271. Should be fixed in the dev branch. If you can't wait for the next release, you could try installing the [JDK](https://jdk.java.net/14/) like I did [here](https://github.com/MarginallyClever/Makelangelo-software/issues/271#issuecomment-640775400).

bjbsquared, 2020-06-25:
It could be. I will get the JDK and try again. If so, I think your assumption holds. For some reason, it was kind of hard to get the link for the installer exe. Google was sending me in all sorts of other places. The link is here: [](https://www.oracle.com/java/technologies/javase-jdk14-downloads.html)

bjbsquared, 2020-06-25:
I am closing this issue because installing the JDK fixed the issue. Refer to #271  for resolution.

closed 2020-06-25 by bjbsquared