MarginallyClever/Makelangelo-software #287: RAMPS Display working but...

jsecondo, 2020-06-26:
I did some research and found the pins for RAMPS config, display is working but the beeper stays on and its driving me crazy.
Tried to set BEEPER to -1 but still turns on. Modified the code that turns on and off the beeper but still it turns on. Any ideas on that?

NOTE: the only pin I have in doubt is SDPOWER 

In board_ramps.h
...
...
// LCD pins
#ifdef LCD_IS_128X64
  // 128x64 full graphics controller
  #define BEEPER             -1//44
  //#define LCD_PINS_RS        19
  //#define LCD_PINS_ENABLE    42
  //#define LCD_PINS_D4        18
  // these alternate pins might help - see https://www.marginallyclever.com/forums/topic/using-reprap-kit-from-amazon-ramps-1-6-lcd-display/#post-22131
  #define LCD_PINS_RS        16
  #define LCD_PINS_ENABLE    17
  #define LCD_PINS_D4        23

  #define LCD_PINS_D5        25//38
  #define LCD_PINS_D6        27//41
  #define LCD_PINS_D7        29//40

  // Encoder rotation values
  #define BTN_EN1            31
  #define BTN_EN2            33
  #define BTN_ENC            37

  // SD card settings
  #define SDPOWER            52//31
  #define SDSS               53//33
  #define SDCARDDETECT       49//35

  #define KILL_PIN           41

/* This comes from : https://forum.arduino.cc/index.php?topic=603486.0
EXP1                        Mega pin

1   beeper                   37
2   encoder switch        35
3   LCD en                   39
4   LCD rs                    38
5   LCD data 4              23
6   LCD data 5              25
7   LCD data 6              27
8   LCD data 7              29
9   ground
10 +5V

EXP2

1   SD MISO                 50
2   SD SCK                   52
3   encoder B               31
4   SD CS                    53
5   encoder A               33
6   SD MOSI                 51
7   SD card detect        49
8   Stop button            41
9   No connect
10  No connect*/
#endif

jsecondo, 2020-06-26:
My bad. The encoder pin was wrong and took the beeper value.

By the way the encoder acts weird. Moves back and forth while trying to go one way or the other.

i-make-robots, 2020-06-26:
The encoders only have four values. Turn faster than the sample rate and it will get confused.

i-make-robots, 2020-06-29:
Please make a pull request to the makelangelo-firmware so that these changes will not be lost.

closed 2020-06-29 by i-make-robots