MarginallyClever/Makelangelo-software #289: Reordered path not efficient

Jip-Hop, 2020-06-26:
Hi Dan,

I've made a test to show the reorder output has issues. The paths in my `reorder_test.svg` are very inefficiently ordered. As you can see by all the pen up moves.

[image]

These are the travel moves when importing with `reorder` checked. Still not much better.

[image]

I imported my `reorder_test.svg` in [saxi](https://github.com/nornagon/saxi), which applied its "sort paths" algorithm (no paths were joined in this case). And the result is much better. The light gray lines indicate pen up moves.

[image]

From saxi I can export the optimized SVG and load it into Makelangelo through this method: https://github.com/nornagon/saxi/issues/49.

In [this zip file](https://github.com/MarginallyClever/Makelangelo-software/files/4837497/reorder.test.svg.files.zip) I've included the original svg and the svg optimized by saxi.

Do you have an idea why the reorder function doesn't seem to work properly with my test file?

i-make-robots, 2020-06-27:
I tried their greedy tour and I don't get the same results.  I know saxi is also removing some lines (identical direction lines merged?) and I wonder if they search for duplicate (same start/end, but flipped) lines.

i-make-robots, 2020-06-27:
updated in 0b0e5fa0

Jip-Hop, 2020-06-27:
Cool it's already in the master branch! Will try it out as soon as possible.

Jip-Hop, 2020-06-27:
Hi Dan,

At first sight reorder is much improved. Way less blue lines.

[image]

But when zooming in, there are still lots of blue lines which follow along black lines.

[image]

Could it be that path reversing isn't applied properly?

By the way, at [line 188](https://github.com/MarginallyClever/Makelangelo-software/blob/0b0e5fa04f00d1ca236dc7263c01c94277d6241f/src/main/com/marginallyclever/artPipeline/ArtPipeline.java#L188) you could use the precomputed `EPSILON2` instead of `EPSILON*EPSILON`.

i-make-robots, 2020-06-28:
Someone is actually reading my code?
https://i.kym-cdn.com/entries/icons/facebook/000/013/306/2dd.jpg

i-make-robots, 2020-06-28:
Yeah, I couldn't figure it out.  I must have spent half a day trying to
reverse things and swap around and reorder... nothing seems to make it as
nice as the example you provided.

Jip-Hop, 2020-06-28:
Thanks Dan. Looks like you're really close! I think I can fix it from here.

Jip-Hop, 2020-06-28:
No more blue lines along black lines!
[image]
Didn't have to change anything about the reorder algorithm. I just wrote the code a bit differently and in the process fixed the error. I think the result of the `flip()` operation wasn't used somehow. 

There's still room for optimizations. Lines are only considered duplicates when they are exactly the same, but not when one line is (partly) inside another line.

Also when drawing there are still more pen-up moves then I feel there should be :P

Jip-Hop, 2020-06-28:
What about the implementation in https://github.com/MarginallyClever/Makelangelo-software/pull/292?

Before
[image]

After
[image]

Before
[image]

After
[image]

Before
[image]

After
[image]

[More SVG test files](https://github.com/MarginallyClever/Makelangelo-software/files/4843066/Archive.zip)

Still room for further optimizations, but at least the greedy reorder now works properly (I hope).

i-make-robots, 2020-06-29:
In the most recent dev branch the reorder_test_simple.svg reorders to 1035 commands instead of the 1831 shown above.  I'd need to set the exact same speeds & feeds to compare the draw time.

Jip-Hop, 2020-06-29:
Well that sounds promising! What about the simple_reorder_test.svg? The reordered path in the screenshot above of that one looks like it can't get much better :p

i-make-robots, 2020-06-29:
I committed your test files as reorder_test*.svg, numbered.  Here are my results:

reorder_test.svg: 7590 instructions, 24m54s.
reorder_test2.svg: 1035 instructions, 3m36s.
reorder_test3.svg: 153 instructions, 2m1s.
reorder_test4.svg: 11 instructions, 0m25s.

ps I am in the company discord if you want to chat

Jip-Hop, 2020-06-29:
My results:

```
Latest dev branch:
reorder_test.svg: 12698 instructions, 0h32m7s.
reorder_test2.svg: 1716 instructions, 0h4m56s.
reorder_test3.svg: 153 instructions, 0h1m46s.
reorder_test4.svg: 11 instructions, 0m15s.
```
```
reorder-fix2 branch:
reorder_test.svg: 13894 instructions, 1h7m26s.
reorder_test2.svg: 1831 instructions, 0h8m13s.
reorder_test3.svg: 150 instructions, 0h1m22s.
reorder_test4.svg: 11 instructions, 0m13s.
```

With reorder_test.svg and reorder_test2.svg you seem to have better results, whereas my results are better with the simpler examples (reorder_test3.svg and reorder_test4.svg). The thing is, reorder_test.svg and reorder_test2.svg have lots of duplicate and overlapping lines. So I think we're not just comparing the reorder performance here, but also other optimizations like deduping. I can clearly see from the "Show pen up moves" in reorder_test3.svg and reorder_test4.svg that my reordering works better...

It's not just the drawing times that aren't comparable with different machine settings it seems. Because I get a lot more instructions compared to you, with the same SVG files and using the same branch.

Jip-Hop, 2020-06-29:
It wasn't the dedupe stage that made things incomparable (we both deduped the same way). It's the higher tolerance for pen-up moves you've introduced with `EPSILON_CONNECTED` which is saving you all those instructions in the messier examples `reorder_test.svg` and `reorder_test2.svg`.

I'm now using the same value your used for `EPSILON_CONNECTED`, and the results are now:

```
Dev branch:
reorder_test.svg: 12698 instructions, 0h32m7s.
reorder_test2.svg: 1716 instructions, 0h4m56s.
reorder_test3.svg: 153 instructions, 0h1m46s.
reorder_test4.svg: 11 instructions, 0m15s.
```

```
reorder-fix2 branch:
reorder_test.svg: 12761 instructions, 0h33m2s.
reorder_test2.svg: 1717 instructions, 0h4m46s.
reorder_test3.svg: 149 instructions, 0h1m21s.
reorder_test4.svg: 11 instructions, 0m13s.
```

I'm still puzzled how yours seems to score better on `reorder_test.svg`, even though the rest of my results are better. I'm not sure if it also results in better draw time though.

I ran the exported gcode through [this tool](https://github.com/Jip-Hop/makelangelo-optimizer) and it reports the following:

```bash
./process.py reorder_test-fix2.ngc > /dev/null
Total Glyphs: 212
Initial penup distance:      6404
Initial total distance:     24188
```

```bash
./process.py reorder_test-dev.ngc > /dev/null
Total Glyphs: 149
Initial penup distance:     30496
Initial total distance:     48113
```

This shows my distance is much shorter (both total and penup), but it still may have more pen up moves...

i-make-robots, 2020-06-29:
Ok, I'm convinced.  Resent the PR and I'll approve it.  In future... more comments always better.  I come back in 6 months and have no idea what I was doing...

i-make-robots, 2020-06-30:
PR approved.

closed 2020-06-30 by i-make-robots