MarginallyClever/Makelangelo-software #293: Crashes on Connect on Windows 10 with JDK14

bjbsquared, 2020-06-29:
Using JDK-14.01 be cause JRE was crashing at startup.

**Heres the top of the file:**

#
# A fatal error has been detected by the Java Runtime Environment:
#
#  EXCEPTION_ACCESS_VIOLATION (0xc0000005) at pc=0x000000007110b5db, pid=18000, tid=16952
#
# JRE version: Java(TM) SE Runtime Environment (14.0.1+7) (build 14.0.1+7)
# Java VM: Java HotSpot(TM) 64-Bit Server VM (14.0.1+7, mixed mode, sharing, tiered, compressed oops, g1 gc, windows-amd64)
# Problematic frame:
# C  [jSSC-2.8_x86_64.dll+0xb5db]
#
# No core dump will be written. Minidumps are not enabled by default on client versions of Windows
#
# If you would like to submit a bug report, please visit:
#   https://bugreport.java.com/bugreport/crash.jsp
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.
#

**Here is the bottom portion of the log file:**
dbghelp: loaded successfully - version: 4.0.5 - missing functions: none
symbol engine: initialized successfully - sym options: 0x614 - pdb path: .;C:\Program Files\Java\jdk-14.0.1\bin;C:\Windows\SYSTEM32;C:\Program Files\AVG\Antivirus;C:\Windows\WinSxS\amd64_microsoft.windows.common-controls_6595b64144ccf1df_6.0.18362.900_none_e6beb9d913147d17;C:\Program Files\Java\jdk-14.0.1\bin\server;C:\Program Files (x86)\Razer\Razer Cortex\x64;C:\Windows\System32\DriverStore\FileRepository\nv_dispi.inf_amd64_d5216eae94436d77;C:\Users\Brian\AppData\Local\Temp\jogamp_0000\file_cache\jln4369785599981916903\jln6794141588261950370\natives\windows-amd64;C:\Users\Brian\.jssc\windows

VM Arguments:
java_command: C:\Users\Brian\Documents\Makelangelo\Makelangelo-7.23.0-with-dependencies.jar
java_class_path (initial): C:\Users\Brian\Documents\Makelangelo\Makelangelo-7.23.0-with-dependencies.jar
Launcher Type: SUN_STANDARD

[Global flags]
     intx CICompilerCount                          = 4                                         {product} {ergonomic}
     uint ConcGCThreads                            = 3                                         {product} {ergonomic}
     uint G1ConcRefinementThreads                  = 10                                        {product} {ergonomic}
   size_t G1HeapRegionSize                         = 1048576                                   {product} {ergonomic}
    uintx GCDrainStackTargetSize                   = 64                                        {product} {ergonomic}
   size_t InitialHeapSize                          = 268435456                                 {product} {ergonomic}
   size_t MarkStackSize                            = 4194304                                   {product} {ergonomic}
   size_t MaxHeapSize                              = 4282384384                                {product} {ergonomic}
   size_t MaxNewSize                               = 2569011200                                {product} {ergonomic}
   size_t MinHeapDeltaBytes                        = 1048576                                   {product} {ergonomic}
   size_t MinHeapSize                              = 8388608                                   {product} {ergonomic}
    uintx NonNMethodCodeHeapSize                   = 5839372                                {pd product} {ergonomic}
    uintx NonProfiledCodeHeapSize                  = 122909434                              {pd product} {ergonomic}
    uintx ProfiledCodeHeapSize                     = 122909434                              {pd product} {ergonomic}
    uintx ReservedCodeCacheSize                    = 251658240                              {pd product} {ergonomic}
     bool SegmentedCodeCache                       = true                                      {product} {ergonomic}
   size_t SoftMaxHeapSize                          = 4282384384                             {manageable} {ergonomic}
     bool UseCompressedClassPointers               = true                                 {lp64_product} {ergonomic}
     bool UseCompressedOops                        = true                                 {lp64_product} {ergonomic}
     bool UseG1GC                                  = true                                      {product} {ergonomic}
     bool UseLargePagesIndividualAllocation        = false                                  {pd product} {ergonomic}

Logging:
Log output configuration:
 #0: stdout all=warning uptime,level,tags
 #1: stderr all=off uptime,level,tags

Environment Variables:
PATH=C:\Program Files (x86)\Razer Chroma SDK\bin;C:\Program Files\Razer Chroma SDK\bin;C:\Program Files (x86)\Razer\ChromaBroadcast\bin;C:\Program Files\Razer\ChromaBroadcast\bin;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Windows\system32\config\systemprofile\AppData\Local\Microsoft\WindowsApps;;C:\Program Files\NVIDIA Corporation\NVIDIA NvDLISR;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Users\Brian\AppData\Local\Programs\Python\Python37-32\Scripts\;C:\Users\Brian\AppData\Local\Programs\Python\Python37-32\;C:\Users\Brian\AppData\Local\Microsoft\WindowsApps;;C:\Users\Brian\AppData\Local\Programs\Microsoft VS Code\bin
USERNAME=Brian
OS=Windows_NT
PROCESSOR_IDENTIFIER=AMD64 Family 23 Model 1 Stepping 1, AuthenticAMD

---------------  S Y S T E M  ---------------

OS: Windows 10 , 64 bit Build 18362 (10.0.18362.900)
OS uptime: 0 days 8:36 hours

CPU:total 12 (initial active 12) (12 cores per cpu, 2 threads per core) family 23 model 1 stepping 1, cmov, cx8, fxsr, mmx, sse, sse2, sse3, ssse3, sse4.1, sse4.2, popcnt, avx, avx2, aes, clmul, mmxext, 3dnowpref, lzcnt, sse4a, ht, tsc, tscinvbit, tscinv, bmi1, bmi2, adx, sha, fma

Memory: 4k page, system-wide physical 16333M (10136M free)
TotalPageFile size 19405M (AvailPageFile size 9119M)
current process WorkingSet (physical memory assigned to process): 179M, peak: 184M
current process commit charge ("private bytes"): 280M, peak: 438M

vm_info: Java HotSpot(TM) 64-Bit Server VM (14.0.1+7) for windows-amd64 JRE (14.0.1+7), built on Mar  5 2020 10:30:14 by "mach5one" with MS VC++ 15.9 (VS2017)

END.

i-make-robots, 2020-06-30:
What happens with the latest development edition?  coz there we don't use JSSC any more, iirc.

bjbsquared, 2020-06-30:
I thought I was running the latest. 
The file name is Makelangelo-7.23.0-with-dependencies.jar. Is this not the latest?

Downloaded from here: 
https://github.com/MarginallyClever/Makelangelo-software/releases/tag/7.23.0

[hs_err_pid10384.log](https://github.com/MarginallyClever/Makelangelo-software/files/4852288/hs_err_pid10384.log)

i-make-robots, 2020-06-30:
Wrong issue number listed.  @bjbsquared, try this: http://www.marginallyclever.com/other/makelangelo%207.23.1.zip

bjbsquared, 2020-06-30:
Nope, it still crashes. 
I can issue GCODE to the machine and get things to happen through the Arduino Serial Interface so I don't think its hardware related. Still related to:
 # Problematic frame:
# C  [jSSC-2.8_x86_64.dll+0xb5db]

**See attached file.** 
[hs_err_pid1460.log](https://github.com/MarginallyClever/Makelangelo-software/files/4854850/hs_err_pid1460.log)

i-make-robots, 2020-07-02:
To help you better, I have spent the last several days learning how to set up instant-build of the dev branch.  Please grab https://github.com/MarginallyClever/Makelangelo-software/releases/download/7.23.0/makelangelo_windows-nightly-20200702-5f2ca4.zip and try it.  If the bug persists, then please paste here the log.txt file generated at run-time.

i-make-robots, 2020-07-03:
This is the latest.  repeat because we were talking in the wrong issue.  https://github.com/MarginallyClever/Makelangelo-software/releases/download/Nightly/makelangelo_windows-nightly-20200703-f8284b.jar

bjbsquared, 2020-07-03:
The crash persists. 

2020-07-03 14:36:54 PROGRAM START
2020-07-03 14:36:54 OS=windows 10
2020-07-03 14:36:54 Locale=en_US
2020-07-03 14:36:54 starting preferences...
2020-07-03 14:36:54 makelangelo.version=7.23.0
2020-07-03 14:36:54 starting robot...
2020-07-03 14:36:54 starting transfer handler...
2020-07-03 14:36:54 starting connection manager...
2020-07-03 14:36:54 starting translator...
2020-07-03 14:36:54 Default language = English
2020-07-03 14:36:54 Looking for translations in jar:file:/C:/Users/Brian/Downloads/makelangelo_windows-nightly-20200703-f8284b.jar!/languages
2020-07-03 14:36:54 rootDir=C:\Users\Brian\Downloads
2020-07-03 14:36:54 Found Jar:languages/piglatin.xml
2020-07-03 14:36:54 Found Jar:languages/english.xml
2020-07-03 14:36:54 Found Jar:languages/dutch.xml
2020-07-03 14:36:54 Creating GUI...
2020-07-03 14:36:54 Create menu bar
2020-07-03 14:36:54   file...
2020-07-03 14:36:54   view...
2020-07-03 14:36:54   help...
2020-07-03 14:36:54   finish...
2020-07-03 14:36:54   adding menu bar...
2020-07-03 14:36:54 create content pane...
2020-07-03 14:36:54   get GL capabilities...
2020-07-03 14:36:56   create draw panel...
2020-07-03 14:36:56   set robot...
2020-07-03 14:36:56   assign panel to robot...
2020-07-03 14:36:56 Cannot run raspistill
2020-07-03 14:36:56   create log panel...
2020-07-03 14:36:56   vertical split...
2020-07-03 14:36:56   horizontal split...
2020-07-03 14:36:56   tweak...
2020-07-03 14:36:56 adjust window size...
2020-07-03 14:36:56   make visible...
2020-07-03 14:36:56   adding drag & drop support...
2020-07-03 14:36:56   starting animator...
2020-07-03 14:36:56 checking sharing permissions...

Heres the error file:
[hs_err_pid21836.log](https://github.com/MarginallyClever/Makelangelo-software/files/4871249/hs_err_pid21836.log)

i-make-robots, 2020-07-05:
Unfortunately the error is inside the native library for JSSC.  As far as I know that library isn't well supported right now.  I'll try to add more messages to find out what happens before that.

Can you open a serial port to the robot with the Arduino app Serial Window?  I want to rule out the chance that windows security is blocking your access.  I use the same JDK so I'm surprised this is happening and unable to reproduce the problem.

bjbsquared, 2020-07-05:
Yes, I am able to communicate and move the robot using Gcode commands with the Arduino serial interface.

i-make-robots, 2020-07-05:
Interesting. Both use jssc.

i-make-robots, 2020-11-10:
I have no updates :(

i-make-robots, 2021-01-02:
Does this help?  https://mcr.dozuki.com/Guide/5.+Makelangelo+Software+Guide/5?lang=en#s357

closed 2021-01-18 by i-make-robots