MarginallyClever/Makelangelo-software #295: Nightly builds by CI for testers that need the dev edition.

i-make-robots, 2020-06-30:
After CI runs and passes,
package final jar into zip file with release contents,
publish release as a daily build,
remove old builds if old > 3 builds.

i-make-robots, 2020-07-02:
Here is the first one!  https://github.com/MarginallyClever/Makelangelo-software/releases/download/7.23.0/makelangelo_windows-nightly-20200702-5f2ca4.zip

As I understand it instant releases will appear in the most recent release, and no more than three instant releases will exist at a time.

bjbsquared, 2020-07-03:
I dont know JAVA or how the IDE works. Looks like there are plugins it can't find:
[image]

i-make-robots, 2020-07-03:
You shouldn't have to run Eclipse if you're using a nightly build.  the zip file *is* a JAR file.  rename it and run it.  I'll see what I can do to keep the .jar name on each nightly build.

i-make-robots, 2020-07-03:
Corrected.  Please try https://github.com/MarginallyClever/Makelangelo-software/releases/download/Nightly/makelangelo_windows-nightly-20200703-f8284b.jar

closed 2020-07-03 by i-make-robots