MarginallyClever/Makelangelo-software #302: Makelangelo does not compile in my place

gsohler, 2020-11-09:
When follow the instructions and import 
"Existing Maven Project"  into eclipse,

it does not compile.
Showing "pom.xml"

it tells: 

Dependencies: Missing artifact net.sourceforge.kabelja:jar:0.4

But looking in the git trunk its there.
same issue happens when trying to compile in the shell terminal:

mvn compile

[ERROR] Failed to execute goal on project Makelangelo: Could not resolve dependencies for project com.marginallyclever:Makelangelo:jar:7.23.0: Failure to find net.sourceforge:kabeja:jar:0.4 in file:///home/gigl/workspace_littleblue/Makelangelo-software/java/local-maven-repo was cached in the local repository, resolution will not be reattempted until the update interval of my-local-repo has elapsed or updates are forced -> [Help 1]

[gigl@computer Makelangelo-software]$ find . -name "kabeja*"
./java/local-maven-repo/org/kabeja
./java/local-maven-repo/org/kabeja/kabeja
./java/local-maven-repo/org/kabeja/kabeja/0.4/kabeja-0.4-sources.jar
./java/local-maven-repo/org/kabeja/kabeja/0.4/kabeja-0.4.jar
./java/local-maven-repo/org/kabeja/kabeja/0.4/kabeja-0.4.pom
./java/local-maven-repo/org/kabeja/kabeja/0.4/kabeja-0.4-javadoc.jar
./local-maven-repo/kabeja-0.4-sources.jar
./local-maven-repo/kabeja-0.4.jar
[gigl@computer Makelangelo-software]$ 

What can be wrong there , i am an absolute maven beginner, so please forgive ...

I use Eclipse version
Eclipse Platform

Version: Oxygen.3a (4.7.3a)
Build id: X20180521-1433

latest version from Maven software(master trunk) as available from 2020-11-09

Thank you for looking into

BTW: Dan , did you notice my reply to your help on Friday 2020-11-06 ?
(only issue with stuck after 1st instructions is solved. home position calucation is still a mystery)

gsohler, 2020-11-10:
Hi Dan,
I have seen that you updated pom.xml . Some new Versions and you added a pluginRepository section in the dev trunk
Also some jar files appeared to arrive. So Eclipse took some time to refresh the workspace, but finally still fails
Please see screenshot.
I also see that the tool has created some directory structure in ~/.m2 ...

[image]

Can you please explain, how this local repository structure should correctly look like and how maven works with the correctly set up directories ?

Thank you

i-make-robots, 2020-11-10:
in dev branch there is a ./local-maven-repo that contains kabeja.  you should be able to point to that to compile.

gsohler, 2020-11-11:
Hi Dan,
I have seen the additional local-maven-repo yesterday already in the dev branch.
Although it was in place, compile did not work out.
I mentioned that i am an absolute beginner with maven, if if there are more steps required  to use the local maven repo, i would definitely fail
Just by tweaking , comparing patterns and reading google, i tried to copy kabeja from
java/local-maven-repo/org
to
~/.m2/repository/org

after that compile works
I do neither know if this is a valid action, or why it needs data outside the Makelanglo branch because i have no clue, whats happening in the background.
If my copy action can be automated or circumvented this will definitely help other users.

Thanks, Guenther

i-make-robots, 2020-11-11:
This is one of those bugs that everybody fixes once and then forgets about.  Is it better to build an automated system OR to add a message in the "getting started" document for developers?

gsohler, 2020-11-11:
Believe that either option is fine because it saves time. But the getting started option is better because scripts are error prone ..see your script with the pushd and popd or the bat file in windows to start makelangelo.
Also it's good if the user knows what happens behind the scene.

i-make-robots, 2020-11-12:
https://github.com/MarginallyClever/Makelangelo-software/wiki/Getting-Started
Better?

gsohler, 2020-11-13:
Yes this is better and can save some time to newbies.
To make the instruction perfect,  please bear in mind that kabeja is a directory , not a jar file at this stage.

Guenther

closed 2020-11-13 by gsohler

i-make-robots, 2020-11-13:
I’m going to delete labels out of .m2 and test again.

gsohler, 2020-11-14:
Hi Dan,

today I completed my first plot successfully.Thank you for sharing.
In some of your Videos i have seen you plotting a photo of a 3D Grid with
steel bars in X, Y and Z direction.
would you mind sharing with me ?

BTW: I have added a pull request which enables to shift the Paper on the
board.

Guenther

i-make-robots, 2020-11-14:
I got the picture here

https://turtletoy.net/turtle/25b7bc4d43

gsohler, 2020-11-14:
Thank you.
This is what I was looking for.
I know these gl programs already.
But I never managed to plot them.
How did you feed to makelangelo?
How to create svg from this?

i-make-robots, 2020-11-14:
Turtletoy will save as svg.  Makelangelo can import svg.

i-make-robots, 2020-11-14:
https://mcr.dozuki.com/Guide/5.+Makelangelo+Software+Guide/5?lang=en#s45

> On Nov 14, 2020, at 3:02 PM, Dan Royer <dan@marginallyclever.com> wrote:
>