MarginallyClever/Makelangelo-software #303: Question about  Custom Makelangelo

gsohler, 2020-11-11:
This is *not* a bug but just a consideration/question.

Me and maybe many other people have built a custom Makelangelo Plotter  which are neither Makelanglo3 nor Makelangelo 5
This means their dimenstions are not well known but arbritrary to the software/firmware.
Fortunately there are forms to enter the Width and height of the plotter.
My Understanding that the Width is the horizontal distance between the pulley centers and Height is the vertical distance
beween the motors and the pen when both timing belts are completley out and their limit switch is hit.
(from my understanding this position is the home position)

Now there is the question:
Assuming I am right and the Home position is the coordinate when both limit switches are hit,
does the Firmware automatically know the timing belt length which is
sqrt(h^2 + W^2/4) 
?
For Firmware to work precisely its important that it knows the exact belt length but i have never seen Makelangelo transferring the belt length. 
Rather i have seen that it transfers the home position which results from the dimensions of my custom plotter and makelangelo5 belt length.

Does Firmware assume/know the correct belt length ?
Would it make sense for makelangelo5  in custom mode to calculate the correct belt length when machine width and height are transmitted ?

AND PLEASE:
I found out by reverse engineering the software that a code of "0"  means Makelangelo Custom. Would be nice to offer that option in the firmware, too ...

i-make-robots, 2020-11-11:
I can confidently say the belt length is not calculated that way right now.  the center of the drawing area is half the height.  the height can be anything.  there was no relation to belts because the design could be more wide than tall, or something weird I've not considered.

There is a D command to set the belt length, iirc.  parser.h details all gcode and their syntax.  Also available in the wiki here.

i-make-robots, 2020-11-12:
D7 [Lnn] [Rnn] - Polargraph only.  Set calibration length of each belt
D8 - Polargraph only.  Report calibration values for left and right belts

gsohler, 2020-11-13:
I use these local changes for the moment but i am aware that this could break official Makalangelo versions ;)

[gigl@computer Makelangelo-software]$ git diff src/main/com/marginallyclever/makelangeloRobot/settings/MakelangeloRobotSettings.java
diff --git a/src/main/com/marginallyclever/makelangeloRobot/settings/MakelangeloRobotSettings.java b/src/main/com/marginallyclever/makelangeloRobot/settings/MakelangeloRobotSettings.java
index d830e8c4..2bb4d458 100644
--- a/src/main/com/marginallyclever/makelangeloRobot/settings/MakelangeloRobotSettings.java
+++ b/src/main/com/marginallyclever/makelangeloRobot/settings/MakelangeloRobotSettings.java
@@ -188,7 +188,8 @@ public double getAcceleration() {
        }
 
        public Point2D getHome() {
-               return getHardwareProperties().getHome();
+//             return getHardwareProperties().getHome();
+               return new Point2D(0,limitBottom);
        }
        /**
         * @return home X coordinate in mm
@@ -210,10 +211,13 @@ public String getGCodeSetPositionAtHome() {
 
        public String getGCodeConfig() {
                String result;
+               double beltlen=Math.sqrt(Math.pow(getHomeX()-limitLeft,2)+Math.pow(limitTop-limitBottom,2));
+               String belt="D7 R"+StringHelper.formatDouble(beltlen)+" L"+StringHelper.formatDouble(beltlen);
+               
                String xAxis = "M101 A0 T"+StringHelper.formatDouble(limitRight)+" B"+StringHelper.formatDouble(limitLeft);
                String yAxis = "M101 A1 T"+StringHelper.formatDouble(limitTop)+" B"+StringHelper.formatDouble(limitBottom);
                String zAxis = "M101 A2 T170 B10";
-               result = xAxis+"\n"+yAxis+"\n"+zAxis; 
+               result = xAxis+"\n"+yAxis+"\n"+zAxis+"\n"+belt; 
                return result;
        }

gsohler, 2020-11-13:
this diff is not correctly displayed, but it can give a rough idea about my local changes.

i-make-robots, 2020-11-23:
at a high level... we need tests to make sure that the app can save & load settings for all types of machines and that the save system will auto-upgrade if new parameters are added in future (instead of forgetting all settings and restarting)

i-make-robots, 2021-01-02:
I added a 

    #define MAKELANGELO_CUSTOM 0 

in the latest master.  It's not used anywhere but it is available and makes the "0 = custom" explicit.