MarginallyClever/Makelangelo-software #312: Custom machine hardware settings are not being saved.

i-make-robots, 2020-11-29:
**Describe the bug**
Hardware settings in firmware EEPROM should be changed as little as possible.  Settings should be applied once and left there.
@gsohler reports custom makelangelo 3s are resetting limits and home position, and on machine start they have to be rechecked every time.

**To Reproduce**
Steps to reproduce the behavior:
1.  install maklenagelo firmware on arduino board
2. connect to software.
3. apply limits and home settings.
4. reboot hardware and software.
5. reconnect and recheck settings.  numbers have changed.

**Expected behavior**
settings should not change between boots.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Platform (please complete the following information):**
 - Device: Arduino-based custom size makelangelos

**Additional context**
see also https://github.com/MarginallyClever/Makelangelo-software/pull/311

gsohler, 2020-11-30:
Hi Just to clarify:
I am author of #311, not #312.
I use D11 command manyally to make sure, that the belt len is saved in the machine.

i-make-robots, 2020-11-30:
@gsohler are you still seeing this issue?

gsohler, 2020-12-01:
I have now connected Makelangelo to my serial terminal and typed 
 D8
to get the belt lengths, and they were correct without defining them.
So it appears they are saved in flash.
Still i feel better that they are set everytime by software(which does happen) . I Dont like to find out a picture looks squeezed just because the belt lengths are wrong by chance.

So to answer your question: No I dont see the issue (anymore).
I did never see a squeezed plot(after i managed to get the microsteps correct)
Still i feel that i very much  want to the belt length to be defined by software.  If i unscrew the motors from my plotter and put them onto a wall, i dont want to miss defining the belt lenght(Width and height is obvious)

Guenther