MarginallyClever/Makelangelo-software #325: Output Exporting issue

sanket0671, 2021-03-25:
I have imported the image and choosed the required conversion style and started to save the file , but the output file given out is not proper file .

expected output file should be of near to  or more than 1mb but the file exported out is of 100kb , which is not proper.

Please help out in this issue so i can get the proper output.

Thanks.

i-make-robots, 2021-03-25:
This might be the worst written bug report I’ve ever seen. No explanation, no steps to repeat, no output file.  We even had a template with instructions!

sanket0671, 2021-03-26:
[image]
this is the Screen shot i can share ,
as you can see the file size reduces to KB's which does not contain any data in it and while exporting it does not give any error also.

i-make-robots, 2021-03-26:
Not repeatable by anyone who can fix this.

sanket0671, 2021-03-26:
what are the possible causes by which this can happen.

i-make-robots, 2021-03-26:
When you opened this ticket you were asked for several things.  You
provided none of them.  I'm not spending time on this until you do the
prerequisites.  if you'd like to read the code yourself and find the
solution on your own then I welcome your pull request.

closed 2021-04-27 by i-make-robots