MarginallyClever/Makelangelo-software #326: I don find the function that moves the sepper motors

edwardocano, 2021-04-12:
I am making a zarplotter, the problem is that i think that the ratio between ropes is not well defined because my motors get more tense when the pen is getting away from the center.  I want to modify the steps. Anyone knows in what file is the function that move the motors to specific step value? 

Thanks for your help!

i-make-robots, 2021-04-12:
instead of changing the step function, change your STEPS_PER_MM or tweak your machine size (top bottom left right limits)
zarplotters are very hard to calibrate and the motors often fight each other.
If you figure out a way to get it right, please let me know.

edwardocano, 2021-04-13:
Where can i change the top bottom left right limits? In the firmware? or in the makelangelo-software?

i-make-robots, 2021-04-13:
You can set the eeprom values by sending a gcode command over serial. The software will do it for you if you like.

edwardocano, 2021-04-13:
Could you send me the link where i can find all these gcodes? Please, also the last firmware version can not connect to mankelangelo 7.24.2. Where can i find specific information about how configure the firmware in platformio? It is a little confuse.

i-make-robots, 2021-04-13:
Sure, here it is.

https://github.com/MarginallyClever/Makelangelo-firmware/wiki/gcode-description

closed 2021-04-27 by i-make-robots