MarginallyClever/Makelangelo-software #329: SVG file incomplete save

martjoy91, 2021-07-17:
Hi. I am using makelangelo 7.24.5 on Windows 7 64x java 1.8.0_291.. 

issue: when i create convert photo to crosshatch (for example), and save it as svg file, and open it in inkscape, there were many lines/paths missing (as per photos).

[image]

[image]

This is my first time using the software. Truly appreciate your kind assistance to help me resolve the issue.

Thank you.

Regards,
mjs

i-make-robots, 2021-07-17:
First may I say this is a well written bug report!  Thank you.  I'll try this with some local images but I may be back to ask for your source image and crosshatch settings.

i-make-robots, 2021-07-17:
Here's my test image loaded with default settings, saved with SVG, and loaded in MS default web browser.  Inkscape 1.0 also loads the SVG looking the same.  Please tell me more so I can recreate the problem.
[image]

martjoy91, 2021-07-18:
> Here's my test image loaded with default settings, saved with SVG, and loaded in MS default web browser. Inkscape 1.0 also loads the SVG looking the same. Please tell me more so I can recreate the problem.
> [image]

Thank you for the super fast response! fyi, the version i was having problem with was downloaded from here:
https://github.com/MarginallyClever/Makelangelo-software/releases/tag/7.24.5

I was using inkscape 0.92.3 earlier. I even opened the SVG from Coreldraw CX7 and were facing similar issue. Because of this, I'm not sure the issue i mentioned earlier still relevant here :) 

On the other hand, earlier today I tried to install one from this (i'm not sure what's the different, I assume the below is the latest version):
https://github.com/MarginallyClever/Makelangelo-software

I followed the step-by-step guide given, installing latest OpenJDK and Eclipse. When i got to this step:

"Errors exist...Proceed with launch?" Select Proceed.

This happened:

[image]

[image]

Was trying to figure out my own way on google, and I'm still stuck. I can't launch the program :(

Perhaps you could advice further?

Thank you so much!

i-make-robots, 2021-07-18:
The module issue is new, cause by me trying to upgrade to java 15. I am actively searching for an answer.

martjoy91, 2021-07-19:
> The module issue is new, cause by me trying to upgrade to java 15. I am actively searching for an answer.
> […](#)

Noted on . The one you share with the photos, what's the makelangelo and java version?

i-make-robots, 2021-07-19:
I ran what was on the master branch, from Eclipse.  I'm not sure why mine works and yours doesn't.  Frankly Eclipse has been weird since I upgraded and if I don't get answers soon I'm going back to Java 8.  At least things were stable there.

i-make-robots, 2021-07-19:
asking for help anywhere I can think of: https://www.reddit.com/r/javahelp/comments/onh89s/upgraded_my_project_from_java_8_to_15_now_jvm/

i-make-robots, 2021-07-19:
I've just made a commit that might fix things.  Please update your copy from github, mavan > update project, and try again.  I look forward to your results.

martjoy91, 2021-07-20:
> I've just made a commit that might fix things. Please update your copy from github, mavan > update project, and try again. I look forward to your results.

Wow. I'm surprised by the effort you took to resolve this issue. really appreciate it. 

I have retried the above, I still have the same issue - 

[image]

I have tried these but still not working:
https://stackoverflow.com/questions/12752064/eclipse-when-launching-in-debug-mode-show-cannot-connect-to-vm-error

i-make-robots, 2021-07-20:
Did you run “maven > update”?
I would also go into debug configurations, delete the existing Makelangelo config, then recreate it.

martjoy91, 2021-07-20:
> Did you run “maven > update”? I would also go into debug configurations, delete the existing Makelangelo config, then recreate it.
> […](#)

Yes i did that. I even went to delete the whole thing, and upload the latest files. still the same thing :(

martjoy91, 2021-07-20:
Anyway, I did went through the different version and gave some of them a try:

7.24.0 - incomplete svg save
7.15.0 - works well

below are the screenshots from 7.15.0

makelangelo:
[image]

Inkspace
[image]

i-make-robots, 2021-07-20:
I want to help and I'm unable to reproduce the error.  What do?

martjoy91, 2021-07-21:
> I want to help and I'm unable to reproduce the error. What do?
> […](#)

For the time being i’ll just use the other version. Will give the latest version a few more trial once in a while. Thank you so much for your help 😄