MarginallyClever/Makelangelo-software #353: SVG Rendering

user01, 2021-10-24:
**Steps to make the bug happen**
1. Load SVG with particular shapes

**What was supposed to happen**
Expected: Render of Gcode to match svg paths
Occurs: Some geometry is not maintained, while some other geometry of similar detail is properly translated

[soduku-mini.svg.txt](https://github.com/MarginallyClever/Makelangelo-software/files/7403953/soduku-mini.svg.txt) (Extension changed for github to allow upload)
[image]
[image]

The malformed 4 outline has the following SVG code:
```xml
<path xmlns="http://www.w3.org/2000/svg" class="cls-1" d="M611.28,1071.74H589.54v-9.81l21.74-25.84h10.4v26.39h5.39v9.26h-5.39v8h-10.4Zm0-9.26V1049l-11.48,13.51Z" transform="translate(-72.31 -483.1)"/>
```

**Platform:**
 - Version: 7.28.3
 - OS: OSX 11.6
 - Robot model: Makelangelo 5

**Log file**

```
2021-10-23 21:54:19 Raspistill unavailable.
2021-10-23 21:54:19 create content pane...
2021-10-23 21:54:19   create PreviewPanel...
2021-10-23 21:54:20   get GL capabilities...
2021-10-23 21:54:20   starting animator...
2021-10-23 21:54:20   make visible...
2021-10-23 21:54:20 adjust window size...
2021-10-23 21:54:20   adding drag & drop support...
2021-10-23 21:54:20 Checking sharing permissions...
2021-10-23 21:54:42 Loading file /Users/erik/soduku-mini.svg...
2021-10-23 21:54:42 Loading...
2021-10-23 21:54:42 ...parse path
2021-10-23 21:54:42 ...parse polylines
2021-10-23 21:54:42 ...parse polygons
2021-10-23 21:54:42 ...parse lines
2021-10-23 21:54:42 ...parse rects
2021-10-23 21:54:42 ...parse circles
2021-10-23 21:54:42 ...parse ellipses
```

**Additional context**
7.28.3 resolved the svg group misalignment

i-make-robots, 2021-12-22:
https://github.com/MarginallyClever/Makelangelo-software/blob/0e0be8537a9ca3b5eec87e390f1e026ad3f56fbd/src/main/com/marginallyclever/makelangelo/makeArt/io/vector/LoadSVG.java#L424 is the bezier rendering method

PPAC37, 2022-02-14:
I confirm, the getNormalizedPathSegList(); is suspicious and the reason for the problem, according to my tests

https://github.com/MarginallyClever/Makelangelo-software/blob/2aae31f77645bfbc23eca47db16297827901819c/src/main/java/com/marginallyclever/makelangelo/makeArt/io/vector/LoadSVG.java#L339-L340

~~~
	    	SVGPathSegList pathList = element.getNormalizedPathSegList();
			outputSVGPathSegList(pathList);
	    	SVGPathSegList pathListNN = element.getPathSegList();
			outputSVGPathSegList(pathListNN);
~~~
~~~

	public void outputSVGPathSegList(SVGPathSegList pathList){
		int size = pathList.getNumberOfItems();
		for ( int i = 0 ; i < size ; i++){			
				SVGPathSeg item = pathList.getItem(i);
				System.out.printf("%s ",((SVGItem)item).getValueAsString());	
		}
		System.out.println();
	}
~~~

give me as output

~~~
M 427.86 1376.92 C 429.52115 1374.0599 431.86407 1371.6552 434.68 1369.92 C 436.85114 1368.504 439.36075 1367.6927 441.94998 1367.5701 C 445.1879 1367.6271 448.24173 1369.0864 450.31998 1371.5701 C 452.7933 1374.2367 454.02664 1378.2935 454.02 1383.7401 C 454.09 1387.7175 453.2485 1391.6582 451.56 1395.2601 C 450.086 1398.448 447.8166 1401.203 444.97 1403.2601 C 441.97415 1405.2946 438.4092 1406.3241 434.79 1406.2001 C 433.61667 1406.2001 431.84668 1406.0801 429.48 1405.8401 C 429.11002 1405.8401 428.57 1405.7401 427.86002 1405.68 L 421.71002 1405.68 L 421.71002 1352.0901 L 427.86002 1352.0901 z M 427.86002 1359.4401 L 427.86002 1375.3401 C 429.88165 1375.87 431.96027 1376.1521 434.05002 1376.18 C 437.6955 1376.2284 441.16525 1374.6167 443.48 1371.8 C 446.04 1368.8734 447.32 1364.8734 447.32 1359.8 C 447.32 1356.1101 446.63 1353.3301 445.23 1351.4601 C 443.83002 1349.5901 442.23 1348.66 440.49002 1348.66 C 438.57 1348.66 436.49002 1349.66 434.13004 1351.5901 C 425.17706 1359.9725 422.6451 1373.1692 427.86 1384.27 z 
M 427.86 1376.92 a 20.05 20.05 0.0 0 1 6.82 -7.0 a 14.57 14.57 0.0 0 1 7.27 -2.35 a 11.17 11.17 0.0 0 1 8.37 4.0 q 3.71 4.0 3.7 12.17 a 26.06 26.06 0.0 0 1 -2.46 11.52 a 20.47 20.47 0.0 0 1 -6.59 8.0 a 17.08 17.08 0.0 0 1 -10.18 2.94 q -1.76 0.0 -5.31 -0.36 c -0.37 0.0 -0.91 -0.1 -1.62 -0.16 h -6.15 v -53.59 h 6.15 z m 0.0 7.35 v 15.9 a 25.77 25.77 0.0 0 0 6.19 0.84 a 12.0 12.0 0.0 0 0 9.43 -4.38 q 3.84 -4.39 3.84 -12.0 c 0.0 -3.69 -0.69 -6.47 -2.09 -8.34 s -3.0 -2.8 -4.74 -2.8 c -1.92 0.0 -4.0 1.0 -6.36 2.93 A 28.29 28.29 0.0 0 0 427.86 1384.27 z 

~~~
and the [first line](https://svg-path-visualizer.netlify.app/#M%20427.86%201376.92%20C%20429.52115%201374.0599%20431.86407%201371.6552%20434.68%201369.92%20C%20436.85114%201368.504%20439.36075%201367.6927%20441.94998%201367.5701%20C%20445.1879%201367.6271%20448.24173%201369.0864%20450.31998%201371.5701%20C%20452.7933%201374.2367%20454.02664%201378.2935%20454.02%201383.7401%20C%20454.09%201387.7175%20453.2485%201391.6582%20451.56%201395.2601%20C%20450.086%201398.448%20447.8166%201401.203%20444.97%201403.2601%20C%20441.97415%201405.2946%20438.4092%201406.3241%20434.79%201406.2001%20C%20433.61667%201406.2001%20431.84668%201406.0801%20429.48%201405.8401%20C%20429.11002%201405.8401%20428.57%201405.7401%20427.86002%201405.68%20L%20421.71002%201405.68%20L%20421.71002%201352.0901%20L%20427.86002%201352.0901%20z%20M%20427.86002%201359.4401%20L%20427.86002%201375.3401%20C%20429.88165%201375.87%20431.96027%201376.1521%20434.05002%201376.18%20C%20437.6955%201376.2284%20441.16525%201374.6167%20443.48%201371.8%20C%20446.04%201368.8734%20447.32%201364.8734%20447.32%201359.8%20C%20447.32%201356.1101%20446.63%201353.3301%20445.23%201351.4601%20C%20443.83002%201349.5901%20442.23%201348.66%20440.49002%201348.66%20C%20438.57%201348.66%20436.49002%201349.66%20434.13004%201351.5901%20C%20425.17706%201359.9725%20422.6451%201373.1692%20427.86%201384.27%20z%20)
[image]

but [the second](https://svg-path-visualizer.netlify.app/#M%20427.86%201376.92%20a%2020.05%2020.05%200.0%200%201%206.82%20-7.0%20a%2014.57%2014.57%200.0%200%201%207.27%20-2.35%20a%2011.17%2011.17%200.0%200%201%208.37%204.0%20q%203.71%204.0%203.7%2012.17%20a%2026.06%2026.06%200.0%200%201%20-2.46%2011.52%20a%2020.47%2020.47%200.0%200%201%20-6.59%208.0%20a%2017.08%2017.08%200.0%200%201%20-10.18%202.94%20q%20-1.76%200.0%20-5.31%20-0.36%20c%20-0.37%200.0%20-0.91%20-0.1%20-1.62%20-0.16%20h%20-6.15%20v%20-53.59%20h%206.15%20z%20m%200.0%207.35%20v%2015.9%20a%2025.77%2025.77%200.0%200%200%206.19%200.84%20a%2012.0%2012.0%200.0%200%200%209.43%20-4.38%20q%203.84%20-4.39%203.84%20-12.0%20c%200.0%20-3.69%20-0.69%20-6.47%20-2.09%20-8.34%20s%20-3.0%20-2.8%20-4.74%20-2.8%20c%20-1.92%200.0%20-4.0%201.0%20-6.36%202.93%20A%2028.29%2028.29%200.0%200%200%20427.86%201384.27%20z%20)
[image]

PPAC37, 2022-02-14:
Also (other issue) , adding a `sinceClosePath=0;` a the end of the case `SVGPathSeg.PATHSEG_CURVETO_CUBIC_ABS`
https://github.com/MarginallyClever/Makelangelo-software/blob/2aae31f77645bfbc23eca47db16297827901819c/src/main/java/com/marginallyclever/makelangelo/makeArt/io/vector/LoadSVG.java#L406
Seems to do the trick for this seahorse (parasite lines disappeared)
[image]
[image]

PPAC37, 2022-02-15:
I don't have the knowledge to implement
~~~
	public static final short PATHSEG_CURVETO_QUADRATIC_ABS = 8;
	public static final short PATHSEG_CURVETO_QUADRATIC_REL = 9;
...
	public static final short PATHSEG_CURVETO_CUBIC_SMOOTH_ABS = 16;
	public static final short PATHSEG_CURVETO_CUBIC_SMOOTH_REL = 17;
	public static final short PATHSEG_CURVETO_QUADRATIC_SMOOTH_ABS = 18;
	public static final short PATHSEG_CURVETO_QUADRATIC_SMOOTH_REL = 19;
~~~
even with a look to ... Generator_Text.writeBeautifulString(Turtle turtle,Font font, FontRenderContext frc,String text,float dx, float dy) ... I'm floundering.

So no way (to my knowledge/skills) to avoid using `getNormalizedPathSegList` ...

i-make-robots, 2022-02-15:
some cleaning in bc9dd294 but nothing has changed.  I have also looked at skipping normalized paths and it is a lot of work.

closed 2022-06-18 by github-actions[bot]