MarginallyClever/Makelangelo-software #354: BADLINENUM with Windows 10

Gregory667, 2021-11-01:
Running the Software under Windows i get a BADLINENUM error from the Makelangelo-Firmware. If i send a command with SPACE char as prefix " N1 G0 X10" it works.

i-make-robots, 2021-11-01:
Makelangelo firmware what version?  We use Marlin as of Makelangelo-software 7.28

Gregory667, 2021-11-01:
I have both. Marlin and Makelangelo firmware. 

Marlin runs fine. But....please correct me if i'm wrong: there is no config for a POLARGRAPH.
-> Pitch of the ballchain or diameter of the belt pulley or distance between the step motors.

So I thought the Makelangelo firmware is best for this.
I use Version 9.2.0

i-make-robots, 2021-11-01:
this version is the one I use for the official firmware sent out with Makelangelo-software. https://github.com/MarginallyClever/Marlin-polargraph/tree/polargraph .  It should be not-awful to adapt it for your needs.  Please refer to the Friday Facts post on our blog about the firmware config process.

closed 2021-11-12 by i-make-robots

Gregory667, 2021-11-13:
Is there a forum or a community for Marlin 2.0 polargraph and makelangelo? 
I need some infos of the position of the steppers in relationship to the bed size.

i-make-robots, 2021-11-13:
There's our Discord.  The Marlin devs don't specifically deal with anything polargraph.
https://discord.gg/2JxQnFPBGA