MarginallyClever/Makelangelo-software #357: Add in pen size control

i-make-robots, 2021-11-08:
Turtle will need a tailDiameter for use in all related calculations.  Part of Turtle tool change.

New pen settings panel in main view will control radius.

Tests:
- open pen settings panel
- save/load settings
- change tailDiameter on tool change (confirm works)
- language changes to the panel
- ??

PPAC37, 2021-12-17:
since https://github.com/MarginallyClever/Makelangelo-software/commit/c1f7e8a342e012e9bf89a5a509d2d3f74aab2b17

 menu bar "Generate art" -> "Fill Page" => GUI freeze (menu list stay visible after the click .

as penDiameter is initialised to 0 and used in a loop as step ...

https://github.com/MarginallyClever/Makelangelo-software/blob/cb56084ab3ba7aa7b9c0ff6e57b1a1707aa6ed49/src/main/com/marginallyclever/makelangelo/makeArt/turtleGenerator/Generator_FillPage.java#L14

https://github.com/MarginallyClever/Makelangelo-software/blob/cb56084ab3ba7aa7b9c0ff6e57b1a1707aa6ed49/src/main/com/marginallyclever/makelangelo/makeArt/turtleGenerator/Generator_FillPage.java#L65

use 0.1 as initialisation value or get the "? tool dimention ?" befor the loop ?

i-make-robots, 2021-12-17:
Whoops.

closed 2022-04-20 by github-actions[bot]