MarginallyClever/Makelangelo-software #359: Importing an .svg with Ellipses turns them into Triangles

GereonPlaettner, 2021-11-13:
**Steps to make the bug happen**
1. Create an ellipse in processing.
2. Export it to .svg
3. Import the .svg into the makelangelo software
4. See error

**What was supposed to happen**
An ellipse should stay elliptical.

**Platform (please complete the following information):**
 - Version: 7.28.2
 - OS: Windows 10
 - Robot model: Makelangelo 5

**Log file**
[log.txt](https://github.com/MarginallyClever/Makelangelo-software/files/7532674/log.txt)

**Additional context**
Using Inkscape to ungroup and convert to path and exporting as dxf12 resolves this in a roudnabout way.

[image]

[filename.zip](https://github.com/MarginallyClever/Makelangelo-software/files/7532671/filename.zip)

i-make-robots, 2021-11-15:
Please try the nightly build and let me know if that's better, too much detail in each circle... your thoughts welcome.

GereonPlaettner, 2021-11-25:
Issue solved for me!

closed 2021-11-25 by GereonPlaettner