MarginallyClever/Makelangelo-software #365: Menu bar "Paper" -> "OpenSettings" = No new jframe and NullPointerException (under Linux Ubuntu 18.04)

PPAC37, 2021-12-16:
Menu bar "Paper" -> "OpenSettings" = No new jframe and NullPointerException (under Linux Ubuntu 18.04)
(same for other JFrames except "PlattersControls" show with no exceptions ...)

java --version

```
java 17 2021-09-14 LTS
Java(TM) SE Runtime Environment (build 17+35-LTS-2724)
Java HotSpot(TM) 64-Bit Server VM (build 17+35-LTS-2724, mixed mode, sharing)

```

~/Downloads/Makelangelo-7.28.2$ java -jar Makelangelo-7.28.2-with-dependencies.jar 
```

log dir=/home/q6/Downloads/Makelangelo-7.28.2/
Exception in thread "AWT-EventQueue-0" java.lang.NullPointerException: Cannot invoke "java.util.ArrayList.add(Object)" because "this.propertyChangeListeners" is null
	at com.marginallyclever.makelangelo.select.SelectPanel.addPropertyChangeListener(SelectPanel.java:71)
	at java.desktop/javax.swing.plaf.synth.SynthPanelUI.installListeners(SynthPanelUI.java:89)
	at java.desktop/javax.swing.plaf.synth.SynthPanelUI.installUI(SynthPanelUI.java:69)
	at java.desktop/javax.swing.JComponent.setUI(JComponent.java:730)
	at java.desktop/javax.swing.JPanel.setUI(JPanel.java:153)
	at java.desktop/javax.swing.JPanel.updateUI(JPanel.java:129)
	at java.desktop/javax.swing.JPanel.<init>(JPanel.java:89)
	at java.desktop/javax.swing.JPanel.<init>(JPanel.java:112)
	at java.desktop/javax.swing.JPanel.<init>(JPanel.java:120)
	at com.marginallyclever.makelangelo.select.SelectPanel.<init>(SelectPanel.java:32)
	at com.marginallyclever.makelangelo.paper.PaperSettings.<init>(PaperSettings.java:61)
	at com.marginallyclever.makelangelo.Makelangelo.openPaperSettings(Makelangelo.java:217)
	at com.marginallyclever.makelangelo.Makelangelo.lambda$2(Makelangelo.java:210)
	at java.desktop/javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:1972)
	at java.desktop/javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2313)
	at java.desktop/javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:405)
	at java.desktop/javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:262)
	at java.desktop/javax.swing.AbstractButton.doClick(AbstractButton.java:374)
	at java.desktop/javax.swing.plaf.basic.BasicMenuItemUI.doClick(BasicMenuItemUI.java:1028)
	at java.desktop/javax.swing.plaf.basic.BasicMenuItemUI$Handler.mouseReleased(BasicMenuItemUI.java:1072)
	at java.desktop/java.awt.Component.processMouseEvent(Component.java:6626)
	at java.desktop/javax.swing.JComponent.processMouseEvent(JComponent.java:3389)
	at java.desktop/java.awt.Component.processEvent(Component.java:6391)
	at java.desktop/java.awt.Container.processEvent(Container.java:2266)
	at java.desktop/java.awt.Component.dispatchEventImpl(Component.java:5001)
	at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2324)
	at java.desktop/java.awt.Component.dispatchEvent(Component.java:4833)
	at java.desktop/java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4948)
	at java.desktop/java.awt.LightweightDispatcher.processMouseEvent(Container.java:4575)
	at java.desktop/java.awt.LightweightDispatcher.dispatchEvent(Container.java:4516)
	at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2310)
	at java.desktop/java.awt.Window.dispatchEventImpl(Window.java:2780)
	at java.desktop/java.awt.Component.dispatchEvent(Component.java:4833)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:773)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:722)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:716)
	at java.base/java.security.AccessController.doPrivileged(AccessController.java:399)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:97)
	at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:746)
	at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:744)
	at java.base/java.security.AccessController.doPrivileged(AccessController.java:399)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86)
	at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:743)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)

```

If i comment in the source code

https://github.com/MarginallyClever/Makelangelo-software/blob/cb56084ab3ba7aa7b9c0ff6e57b1a1707aa6ed49/src/main/com/marginallyclever/makelangelo/Makelangelo.java#L787

or 

https://github.com/MarginallyClever/Makelangelo-software/blob/cb56084ab3ba7aa7b9c0ff6e57b1a1707aa6ed49/src/main/com/marginallyclever/makelangelo/Makelangelo.java#L170

and run it under NetBeans no more exceptions and the new JFrame "PaperSettings" show.

Or i may have Java LookAndFeel config to do ?
Have a good day.

i-make-robots, 2021-12-16:
Neither of the lines you mention has anything to do with PaperSettings.  I think you are correct, you have look & feel config issue.

PPAC37, 2021-12-16:
Thank you ... unfortunately I don't know how to configure the Java LookAndFeel (I have some documentation to find / read).

But strangely in my System / Java environment it seems that this method (@Override of PropertyChangeListener that this class implements),

https://github.com/MarginallyClever/Makelangelo-software/blob/cb56084ab3ba7aa7b9c0ff6e57b1a1707aa6ed49/src/main/com/marginallyclever/makelangelo/select/SelectPanel.java#L70-L72

happens to be called before the instantiation of the class and therefore to put "static"

https://github.com/MarginallyClever/Makelangelo-software/blob/cb56084ab3ba7aa7b9c0ff6e57b1a1707aa6ed49/src/main/com/marginallyclever/makelangelo/select/SelectPanel.java#L68

corrects the problem for me because it is then no longer null.
But I don't know if it's okay to pass this to static or if there's something I'm missing.

Thank you.

i-make-robots, 2021-12-17:
propertyChangeListener should not be static.  Static makes the list shared by all panels at the same time, leading to unintended side effects.

PPAC37, 2021-12-17:
Sorry ... I edited my post before your reply when I realized my mistake. (My initial proposal to set static propertyChangeListener was totaly wrong !)

And I do not know if to check if propertyChangeListener is not null before using its method .add (...) , is or not a good solution.

i-make-robots, 2021-12-17:
Idk how it can be null. It’s created before the constructor is called. It’s never destroyed. Should be impossible.

PPAC37, 2021-12-17:
I also don't really understand ...
but the SelectPanel constructor via super(); triggers a chain of initialization calls (Jpanel, Jcomponent, ...) and (on my system for some LookAndFeel) something do a installListeners before the full instantiation of the class SelectPanel is done.

PPAC37, 2021-12-19:
Specifying a usable java "Look And Feel" to the JVM is not very "user friendly"/"java beginner accesible" so i reopen this ticket.

i-make-robots, 2021-12-19:
in b30911ee and 2bf0965b I removed all UIManager.setLookAndFeel from every test in the code and made it optional in the jar file.  Use command line option "-nolf" to disable the call.

i-make-robots, 2021-12-19:
maybe 65007b5e helps?  I wonder if moving the declaration for the propertyChangeListeners allows it to be created before super() is called.  Can you confirm if that works?

PPAC37, 2021-12-19:
Thank you !

Using "-nolf" as an argument works fine and is more convenient for me than using "-Dswing.systemlaf = ...".

But unfortunately
https://github.com/MarginallyClever/Makelangelo-software/commit/65007b5e631eca7b74d8d9f8803c825eed056b63 does not fix the problem (that -nolf allows to work around for me.)
```
Exception in thread "AWT-EventQueue-0" java.lang.NullPointerException: Cannot invoke "java.util.ArrayList.add(Object)" because "this.propertyChangeListeners" is null
	at com.marginallyclever.makelangelo.select.SelectPanel.addPropertyChangeListener(SelectPanel.java:69)
	at java.desktop/javax.swing.plaf.synth.SynthPanelUI.installListeners(SynthPanelUI.java:89)
```

I still don't understand the reason ... my system (OS, JVM) or Threads issue (SwingWorker vs Main) or something else ...

Maybe put a try catch so that the Swing Worker keep working ... and make an entry in the Log (with the LookAndFeel class name in use) that talks about testing "-nolf"  ...

i-make-robots, 2021-12-20:
9996a26e tries again.  Any better?

PPAC37, 2021-12-20:
Yes ! 
https://github.com/MarginallyClever/Makelangelo-software/commit/9996a26eb2d9f13e2ddb478ec5b549f2bd4ac639 works perfectly and solved my issue !

Many thanks and well done!

closed 2021-12-20 by PPAC37