MarginallyClever/Makelangelo-software #367: PaperSettings : "⇋" ("Paper orientation/landscap") CheckBox do not work properly.

PPAC37, 2021-12-19:
**Steps to make the bug happen**

Try to change the paper orientation (clic on the "⇋" CheckBox) in Paper settings 
(not with a paper size "---" but "A2" for exemple)

**What was supposed to happen**

paper w and h permuted and view update.

But

CheckBox "flicker", and no orientation change ( paper width and height not permuted, no changes in view)
(But ok with a "---" custom different paper size ( width, height different from other selectables papers size))

**Platform (please complete the following information):**
 - Version: 7.28.2
 - OS: Linux Ubuntu 18.04

**Log file**
(nothing usefull for me ...)
[log.txt](https://github.com/MarginallyClever/Makelangelo-software/files/7740536/log.txt)

PPAC37, 2021-12-19:
If i comment this 
https://github.com/MarginallyClever/Makelangelo-software/blob/cb56084ab3ba7aa7b9c0ff6e57b1a1707aa6ed49/src/main/com/marginallyclever/makelangelo/select/SelectBoolean.java#L43
my build doesn't have this problem. But maybe it is full for a specific JVM version.

coliss86, 2021-12-20:
I have the same issue.

mac os  11.6
```
❯ java  -version
openjdk version "17" 2021-09-14
OpenJDK Runtime Environment Temurin-17+35 (build 17+35)
OpenJDK 64-Bit Server VM Temurin-17+35 (build 17+35, mixed mode, sharing)
```

PPAC37, 2021-12-20:
Well done ! 
To me, with this fix, this issue seems to be resolved.

Thanks.

closed 2021-12-21 by PPAC37