MarginallyClever/Makelangelo-software #368: Margin / Border and therfore "Outline paper" not paper centered if PaperSettings have a Shift X and/or Y.

PPAC37, 2021-12-19:
**Steps to make the bug happen**
Papper Settings, use a Shit X 

**What was supposed to happen**
Margin centered on paper ... but not.
And this inpacte "Outline paper" and "Paper fill" (but Paper fill have also other problèms .. see screenshot 2)

**Platform (please complete the following information):**
 - Version: 7.28.2
 - OS: linux Ubuntu 18.04

**Log file**
(Nothing usefull for me in it ...)
[log.txt](https://github.com/MarginallyClever/Makelangelo-software/files/7740589/log.txt)

**Additional context**
[image]
[image]

PPAC37, 2021-12-19:
https://github.com/MarginallyClever/Makelangelo-software/blob/cb56084ab3ba7aa7b9c0ff6e57b1a1707aa6ed49/src/main/com/marginallyclever/makelangelo/paper/Paper.java#L173-L204

Have to take in account the page center position (due to the shift ) ?
Somethign like this 

```
	public double getMarginRight() {
		//return getPaperRight() * getPaperMargin();
                   double centrerCoord = (getPaperRight()+getPaperLeft()) / 2;		  
                return ((getPaperRight() - centrerCoord )* getPaperMargin() ) +centrerCoord ;                
	}
```
But not applicable if rotation (that also dont seem to work) is used.

i-make-robots, 2021-12-19:
maybe it's worthwhile to have a getCenterX() and a getCenterY()?  i can see 

    double centrerCoord = (getPaperRight()+getPaperLeft()) / 2;	

being used in at least three other places.

i-make-robots, 2021-12-19:
Added them in ccb16fdd and 6f9dc357.  Appears to adjust paper size on-the-fly as I change shiftX and shiftY in the paper settings menu.  Margin also tested.  Rotation doesn't appear to do anything.

PPAC37, 2021-12-20:
Thanks
Perfect rendering in the view. The margin is well centered on the sheet even when shifted.

I wonder whether or not the size of the pen should be taken into account when drawing. It seems to me that if we trace the margin, then it extends over half the size of the pen. ... but I have never used large pens (> 0.5mm) ... so I don't really know.

For on-the-fly rendering it is surely thanks to the update of myPaper via the listeners of the Shift X and Shit Y fields.
I added in a test an "updatePaperFromPanel ();" at the end of "onPaperDimensionsChange (PropertyChangeEvent e)" of the PaperSettings class to have the same thing when modifying the custom PaperSize dimensions ("---") and it seems to work.

For me, this specific problem of margin position is solved. 
But it raises other questions and maybe problems. (Rotation? Paper Fill ? Margin Overflow?)

i-make-robots, 2021-12-20:
True!  We may need to start a separate Discussion thread about these options.  Rotation I threw in, not sure it was ever needed.  there's no way to set negative margin.  One day it would be nice to have a "split picture into n pieces" so they tile to make larger art.

closed 2021-12-20 by i-make-robots