MarginallyClever/Makelangelo-software #383: PNG Image do not load

coliss86, 2021-12-21:
**Steps to make the bug happen**
1. File > Open File
2. Select a png
3. an empty dialog with an error title is displayed and the application is frozen
[image]

**What was supposed to happen**
The image should be loaded

**Platform (please complete the following information):**
 - Version: 7.30.2
 - OS: macos 11.6

**Log file**
[log.txt](https://github.com/MarginallyClever/Makelangelo-software/files/7757060/log.txt)

**Additional context**
loading JPEG also failed

i-make-robots, 2021-12-21:
Confirmed.  In the error message I get

    Cannot invoke "com.marginallyclever.makelangelo.makeArt.TransformedImage.canSampleAt(double,double)" because "this.myImage" is null

Which is very cryptic.  It looks to me like a failure inside ConvertImagePanel.

i-make-robots, 2021-12-21:
On a separate note, "Missing: Error" is silly.  Your system shows blank for `e.getLocalizedMessage()` while the log file probably shows the message correctly with `"Load error:"+e.getMessage()`

coliss86, 2021-12-21:
I have added a `printStackTrace()` : 
```
java.lang.NullPointerException: Cannot invoke "com.marginallyclever.makelangelo.makeArt.TransformedImage.canSampleAt(double, double)" because "this.myImage" is null
	at com.marginallyclever.makelangelo.makeArt.imageConverter.Converter_VoronoiStippling.initializeCells(Converter_VoronoiStippling.java:333)
	at com.marginallyclever.makelangelo.makeArt.imageConverter.Converter_VoronoiStippling.restart(Converter_VoronoiStippling.java:209)
	at com.marginallyclever.makelangelo.makeArt.imageConverter.Converter_VoronoiStippling.propertyChange(Converter_VoronoiStippling.java:177)
	at java.desktop/java.beans.PropertyChangeSupport.fire(PropertyChangeSupport.java:343)
	at java.desktop/java.beans.PropertyChangeSupport.firePropertyChange(PropertyChangeSupport.java:335)
	at java.desktop/java.beans.PropertyChangeSupport.firePropertyChange(PropertyChangeSupport.java:268)
	at java.desktop/java.awt.Component.firePropertyChange(Component.java:8722)
	at java.desktop/java.awt.Component.updateGraphicsData(Component.java:1187)
	at java.desktop/java.awt.Container.updateChildGraphicsData(Container.java:1187)
	at java.desktop/java.awt.Component.updateGraphicsData(Component.java:1179)
	at java.desktop/java.awt.Container.updateChildGraphicsData(Container.java:1187)
	at java.desktop/java.awt.Component.updateGraphicsData(Component.java:1179)
	at java.desktop/java.awt.Component.setGraphicsConfiguration(Component.java:1167)
	at java.desktop/java.awt.Container.addImpl(Container.java:1140)
	at java.desktop/java.awt.Container.add(Container.java:440)
	at com.marginallyclever.makelangelo.makeArt.io.LoadFilePanel.load(LoadFilePanel.java:88)
	at com.marginallyclever.makelangelo.makeArt.io.LoadFilePanel.chooseFile(LoadFilePanel.java:74)
	at com.marginallyclever.makelangelo.makeArt.io.LoadFilePanel.lambda$getFileSelectionPanel$0(LoadFilePanel.java:60)
	at java.desktop/javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:1972)
	at java.desktop/javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2313)
	at java.desktop/javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:405)
	at java.desktop/javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:262)
	at java.desktop/javax.swing.plaf.basic.BasicButtonListener.mouseReleased(BasicButtonListener.java:279)
	at java.desktop/java.awt.Component.processMouseEvent(Component.java:6626)
	at java.desktop/javax.swing.JComponent.processMouseEvent(JComponent.java:3389)
	at java.desktop/java.awt.Component.processEvent(Component.java:6391)
	at java.desktop/java.awt.Container.processEvent(Container.java:2266)
	at java.desktop/java.awt.Component.dispatchEventImpl(Component.java:5001)
	at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2324)
	at java.desktop/java.awt.Component.dispatchEvent(Component.java:4833)
	at java.desktop/java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4948)
	at java.desktop/java.awt.LightweightDispatcher.processMouseEvent(Container.java:4575)
	at java.desktop/java.awt.LightweightDispatcher.dispatchEvent(Container.java:4516)
	at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2310)
	at java.desktop/java.awt.Window.dispatchEventImpl(Window.java:2780)
	at java.desktop/java.awt.Component.dispatchEvent(Component.java:4833)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:773)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:722)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:716)
	at java.base/java.security.AccessController.doPrivileged(AccessController.java:399)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:97)
	at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:746)
	at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:744)
	at java.base/java.security.AccessController.doPrivileged(AccessController.java:399)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86)
	at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:743)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)
```

i-make-robots, 2021-12-21:
Converter_VoronoiStippling.initializeCells() is probably the guilty party.

i-make-robots, 2021-12-21:
The idenpendent test (run ImageConverterPanel as a java application) crashes with error message immediately.
Fixed that and now the app runs up to the conversion options but then several options do not have cause-and-effect.  eg i change the config options inside crosshatch and nothing changes on screen because the converter does not restart.

A lot of this code is spaghetti because it tried to work nice with single-pass converters like cross hatch and iterative converters like voronoi stippling.

i-make-robots, 2021-12-21:
b5d9be1c should fix everything.

i-make-robots, 2021-12-21:
It would be awesome if we had an AssertJ unit test for this behavior to make sure it never happens again.

coliss86, 2021-12-21:
Fix confirmed !

coliss86, 2021-12-21:
Can you tell me what did you run to reproduced the problem ? I can try to implement is as a test

i-make-robots, 2021-12-21:
I followed your steps, as described in the OP.

closed 2021-12-21 by i-make-robots

PPAC37, 2021-12-21:
Oops ... my apologies! the problem from here was related to the fixes to fix my LookAndFeel problem
https://github.com/MarginallyClever/Makelangelo-software/issues/365
and with this last fix my problem and come back.

But I have the option of using -nolf to bypass it ... so I'll leave it at that for now. (Mom problem and closed because I can bypass it and this one too.)

coliss86, 2021-12-21:
> I followed your steps, as described in the OP.

You said previously : "The idenpendent test (run ImageConverterPanel as a java application) crashes with error message immediately." but I can't find any runnable method. May be I didn't understood what you mean.