MarginallyClever/Makelangelo-software #398: Log and LogBack

coliss86, 2021-12-23:
I don't understand why LogBack is used in `com.marginallyclever.convenience.log.Log` in combination a manual write in a log file ?
My understanding is that is used as a crash reporter and the user can send to you the log of the previous execution.
I have also see that it is used to append log in dialog box of Marlin with a listener.

When writing a log with the a logger framework, you only need to declare the logger once in a class and used it. The logger do all the heavy stuff like opening a file, writing in it, checking the level of log etc...
`com.marginallyclever.convenience.log.Log` reduces the API of logback and for example when an exception occurs [like here](https://github.com/MarginallyClever/Makelangelo-software/blob/master/src/main/com/marginallyclever/util/MarginallyCleverPreferencesHelper.java#L46), the log does not have a full stack trace neither the root cause of the problem. It lacks information which can be very useful for debugging.
With a logger, you can write something like : 
```
String myFile = "/tmp/toto";
try {
..
} catch (IOException e) {
    logger.error("Error while opening the file {}", file, e);
}
```
And you get a line containing "Error while opening the file /tmp/toto: IOException....."
And a full stacktrace with all the details. 

As a new developer on Makelangelo-software, the lack of log make the debugging hard. You'll tell me to look for the `log.txt` file but in dev mode, all the log should go to the stdout in the IDE. Moreover, there is no "debug" or "info" log, and it makes harder to understand what is really useful or less.

I would like to propose to you the rewrite of this log class by using sl4f + logback in order to have a `log.txt` file when running the app in standalone mode and with more detailed info when debugging it.
Before starting this and to avoid the pain in PR #369 where all the code is going to the trash, are you OK with such rewrite? What are your expectations ? Do you want to throw away something or do you have something else in plan ?

Regards

i-make-robots, 2021-12-23:
Legacy stuff I didn't write.

i-make-robots, 2021-12-23:
as long as you don't use log4j! lol

coliss86, 2021-12-23:
> as long as you don't use log4j! lol
🤣

Issues has been found in log4j, tomorrow issue in another heavily used lib will be discovered... ;)

PR #402 (2021-12-25): DRAFT: Rewrite Log to use sl4fj + logback instead
Fix #398

New LogPanel with a copy to clipboard button:
[image]

DO NOT MERGE, it is not finished

Left to do:

- [x] move all call from Log.message() to logger.debug()
  - [x] firmwareUploader
  - [x] makeArt
  - [x] makelangeloSettingsPanel
  - [x] paper
  - [x] plotter
  - [x] preview
  - [x] select
  - [x] turtle
- [x] check all catch Exception to add more info
- [x] check if the CI is OK
- [x] rewrite LogPanel
closed 2021-12-25 by i-make-robots