MarginallyClever/Makelangelo-software #399: PlotterControls Panel

coliss86, 2021-12-23:
I think the PlotterControls needs a lot of click to launch a drawing: 
[image]

You have to click to MarlinInterface, connect, wait for the list of available device to display, select your device (which is the third result for me), click OK, click on the little button "Find home" and next to "Draw".
Here is a some things I propose: 
- The latest selected device (COM2, /dev/cu...) is selected by default and the dropdown menu is placed directly in the main panel
- make the toolbar with "Find home" and other buttons bigger with some icons
- clicking on the MarlinInterface should only be required if you want to see the gcode. On my computer (mac os 11.6, 2,2 GHz 6-Core Intel Core i7, 16 Gb of RAM), on some drawing, the log of the gcode slow down the robot. If I switch to JogInterface during the print, the robot is smoother. This is due to the log panel which is not fast enough to display a large amount of info. I do not think is it a bug of this software but a limitation of Swing.
- when clicking connect, if the printer is not ready or absent, there is not error message: 
[image]

What do you think ? Do you have something in mind that I should be aware of ?

Regards

i-make-robots, 2021-12-23:
I'm glad we're discussing this before writing code.  My primary goal with this was to write each component to be independently testable.  

I see no conflict with font size changes, I think remembering last COM port used is a Good Thing.  Same goes for last paper size chosen.

PlotterControls has an instance of JogInterface, ProgramInterface, and MarlinPlotterInterface.  Therefore PlotterControls can have a button that calls MarlinPlotterInterface.OnConnect (sic).  I like that because it means 99% of people will never have to see the scary MarlinPlotterInterface or the ProgramInterface.

> when clicking connect, if the printer is not ready or absent, there is not error message:

If I click "connect" and there are no COM ports available, the list is empty.  Can you describe the steps more clearly, please?  Leaving the user with a mystery is a kind of bug, imho.

coliss86, 2021-12-23:
> If I click "connect" and there are no COM ports available, the list is empty. Can you describe the steps more clearly, please? Leaving the user with a mystery is a kind of bug, imho.

On macos, there is always `/dev/cu.Bluetooth-Incoming-Port` in the list of port even if no robot is connected or even when the bluetooth is switched off (as its name implied).
If I connect to the wrong port, no error is reported, I do not think the output of Marlin is parsed to look for a pattern.
`/dev/cu.Bluetooth-Incoming-Port` is a valid serial interface but not Marlin compatible.
The 2nd screenshot upper is a example of this behaviour.

coliss86, 2022-01-07:
Close by #416

closed 2022-01-07 by coliss86