MarginallyClever/Makelangelo-software #404: FirmwareUploader will not work on linux ou macos

coliss86, 2021-12-24:
In `FirmwareUploader`, the path to avrdude is hard coded for windows even if the binary file name is calculated for windows or *nix.
cf : https://github.com/MarginallyClever/Makelangelo-software/blob/6dde11ce053c650c6a6b4de8d4d92e6a9d531d9f/src/main/java/com/marginallyclever/makelangelo/firmwareUploader/FirmwareUploader.java#L28

On linux ou macos, avrdude should be in the path, no need to specify a folder.

PPAC37, 2021-12-24:
Hello !
Sorry for not agreeing, when I look at the context I think it works ... but I haven't tested.

https://github.com/MarginallyClever/Makelangelo-software/blob/6dde11ce053c650c6a6b4de8d4d92e6a9d531d9f/src/main/java/com/marginallyclever/makelangelo/firmwareUploader/FirmwareUploader.java#L16-L38

But a hard-coded path is a bad practice especially since all Windows versions do not necessarily use the same path for% ProgramFiles%,% ProgramFiles (x86)%,% CommunProgramFiles%,% CommunProgramFiles (x86)% .. .

For me, it's rather the non-management of the process return code that bothers me. Normally a program which exits without error returns 0?
https://github.com/MarginallyClever/Makelangelo-software/blob/6dde11ce053c650c6a6b4de8d4d92e6a9d531d9f/src/main/java/com/marginallyclever/makelangelo/firmwareUploader/FirmwareUploader.java#L71-L75

i-make-robots, 2021-12-25:
It's a fairly new tool and i agree there is much here to dislike.  The output of the AVRDude is not echoed in real time so the user has no idea when the job is actually done or if it is working.  Even a progress bar would be better.  I had difficulty getting the stdout from avrdude while it was running, which is to say that I tried already and did not yet succeed.

PPAC37, 2021-12-25:
See https://github.com/MarginallyClever/Makelangelo-software/pull/405

PPAC37, 2022-02-25:
I'm sorry, I'm dragging my feet for this PR... I'm cleaning up soon... and out of draft ...

closed 2022-05-05 by github-actions[bot]