MarginallyClever/Makelangelo-software #408: Log.txt not found

PPAC37, 2021-12-26:
Something i dont get ... / do wrong ?

**Steps to make the bug happen**
get the nightly build, run it in a fresh sub dir,
```
$ echo $DEV

$ java -jar makelangelo-nightly-20211225-69761a.jar -nolf

```
 try to look the Log panel. i only found a  
```
Failed to read log file: /home/q6/Downloads/log.txt (Aucun fichier ou dossier de ce type)
```

**Second user case (env variable "DEV" set to "true")**
```
$ export DEV=true
$ echo $DEV
true
$ java -jar makelangelo-nightly-20211225-69761a.jar -nolf
2021-12-26 04:10:05,506 DEBUG c.m.convenience.CommandLineOptions - START OPTION -nolf 
2021-12-26 04:10:05,510 DEBUG c.m.makelangelo.Translator - starting translator... 
2021-12-26 04:10:05,535 DEBUG c.m.makelangelo.Translator - Default language = French 
...
2021-12-26 04:10:11,914 DEBUG c.m.makelangelo.Makelangelo -   adding drag & drop support... 
2021-12-26 04:10:20,210 ERROR c.m.convenience.log.LogPanel - Failed to read log file /home/q6/Downloads/log.txt 
java.io.FileNotFoundException: /home/q6/Downloads/log.txt (Aucun fichier ou dossier de ce type)
	at java.base/java.io.FileInputStream.open0(Native Method)
	at java.base/java.io.FileInputStream.open(FileInputStream.java:216)
	at java.base/java.io.FileInputStream.<init>(FileInputStream.java:157)
	at java.base/java.io.FileReader.<init>(FileReader.java:75)
	at com.marginallyclever.convenience.log.LogPanel.<init>(LogPanel.java:30)
	at com.marginallyclever.convenience.log.LogPanel.runAsDialog(LogPanel.java:54)
	at com.marginallyclever.makelangelo.Makelangelo.runLogPanel(Makelangelo.java:560)

```

Linux Ubuntu 18.04

coliss86, 2021-12-26:
I'm looking at it

PPAC37, 2021-12-26:
Thanks.

I tested on a Win7 no problem.

It is surely still a fishy thing specific to my environment, my Ubuntu 18.04 is rather polluted.

Don't waste too much time on it.
I'll see later with a refresh up-to-date linux Ubuntu 20.40 install.

coliss86, 2021-12-26:
there is a bug, thanks for testing ! <3

coliss86, 2021-12-26:
Wait before merging, I have trouble with the CI

closed 2021-12-26 by i-make-robots

PPAC37, 2021-12-26:
Well done !  Thanks.

commit 0542c4dd (2021-12-26): Merge pull request #410 from coliss86/fixlog

Fix #408: improve log rotation and crash detection
commit f0ceec7b (2021-12-26): Fix #408: improve log rotation and crash detection