MarginallyClever/Makelangelo-software #412: Preferences load and save

coliss86, 2021-12-26:
Hi, 

I tried to understand how `MarginallyCleverPreferences` works.
I started from `PreferencesHelperTest` but I did not understand.

It seems over-complicated to save some preferences.
Can you explain to me how it works please ?

PPAC37, 2021-12-26:
I still have to understand it well.

But from what I understand, we have a model with knots. (a preference tree therefore sub-groups)

It is loaded at start-up. By the class "PreferencesHelper"
to access an element, you have to ask for a node

```
Preferences paperPreferenceNode =
PreferencesHelper.getPreferenceNode (PreferencesHelper.MakelangeloPreferenceKey.PAPER);
```
and interleave the attributes of this node.
```
paperPreferenceNode.get ("AttributKeyName" [, default value]);
```
if the attributes of the node are modified, the modifications are not written directly but are updated in the node.
```
paperPreferenceNode.putDouble ("AttributKeyName", someValue);
```
it is necessary to go through the normal end of the program which implements the writing of the files whether or not there has been modification.

It seems to me that it is effective for later adding a notion of vetoable (view it a way to implement undo / "ctrl + Z" on prefrences.)

i-make-robots, 2021-12-26:
I didn't write it, I haven't touched it.  Peter C... wrote it many years ago.  I'd be fine with using a local text file that's easy to find.

coliss86, 2021-12-26:
Thx both for your answer, I'll propose a new implementation with a properties files

PPAC37, 2021-12-26:
I do not have a good perception of the advantage / disadvantage of one or the other.
(once we have made some customary utility class, we should be able to have the same kind of code to enter to add / modify / read / write groups / attributes.

I would say if you do it (seems a relatively big job) I couldn't refuse it. but I don't have the right to write to this github;).

Do you think ini4j? or really the basic / simple properties file?

coliss86, 2021-12-26:
I'm thinking of a simple properties file because for now, there well be at most 20 key / values to store. Even with 40 key, it will be usable.
If later something more complex is necessary, it could be changed.

i-make-robots, 2021-12-27:
imho current preference system isn't broken.  only the action to USE it is being dumb.  let's fix that and focus on more important stuff.

closed 2021-12-29 by i-make-robots