MarginallyClever/Makelangelo-software #415: QOI image format support?

i-make-robots, 2021-12-26:
**Is your feature request related to a problem? Please describe.**
Add support for new QUI image format

**Describe the solution you'd like**
see https://github.com/saharNooby/qoi-java

**Describe alternatives you've considered**
None

**Additional context**
https://tech.slashdot.org/story/21/12/25/0618218/quite-ok-image-format-qoi-coming-to-a-graphics-program-near-you

PPAC37, 2021-12-26:
interesting, for me it's new, ... so we'll have to play with it;)

i-make-robots, 2021-12-26:
tested with https://github.com/saharNooby/qoi-java/blob/main/src/test/resources/dice.qoi :
[image]

closed 2021-12-26 by i-make-robots

PPAC37, 2021-12-26:
Well done !