MarginallyClever/Makelangelo-software #417: Strange behavior of SelectTextArea in the SelectPanel on a window resize.

PPAC37, 2021-12-27:
To summarize the good use of class Select* still escapes me. 

Is there an example with a SelectTextArea ( a Select containing a JTextArea in a JScrollPane ) that doesn't have a strange behavior?

Strange behavior of SelectTextArea in the SelectPanel when resizing the SelectPanel : 
Menu bar -> "Generate art" -> "Your message here" and try to resize multiple times the window "Your message here".

i-make-robots, 2021-12-27:
Good catch, @PPAC37 !

closed 2021-12-27 by i-make-robots

PPAC37, 2021-12-27:
Well done!
I was completely lost on the source of the problem. and therefore the solution. And you correct that, in less than 10 minutes.

i-make-robots, 2021-12-27:
It never hurts to ask the guy who's been reading the code since day 1 :)

commit 3ab64973 (2021-12-27): fixed #417