MarginallyClever/Makelangelo-software #418: Since the adaptation to match the maven recommentation src dir structure the javadoc generated is kind of empty.

PPAC37, 2021-12-27:
Im not used to maven so this may by mormal ... or i'm not doing it the correct way (generating the javadoc) 

But the generated javadoc is kind empty ... no class detail, only "inheritance graph".

i-make-robots, 2021-12-27:
I'm dumb - what's the maven goal to generate javadoc?

i-make-robots, 2021-12-27:
I see maven-javadoc-plugin is commented out in my pom.xml

PPAC37, 2021-12-27:
i dont use maven goals to generat the javadoc but under Netbeans IDE there is a "generat javadoc" on the contextual menu of a project.

PPAC37, 2021-12-27:
> I see maven-javadoc-plugin is commented out in my pom.xml

no maven goal for javadoc ... maybe it's done intentionally ... A way of saying "javadoc is for the weak", you have to look at the source code! ;)

i-make-robots, 2021-12-27:
it was probably me commenting it out because it regenerated every time i ran the "package" goal.  I don't want it all the time, thank you!  It's embarrassing how little is documented, thank you!

i-make-robots, 2021-12-29:
I'm running Eclipse > Project > Generate Javadoc and I get a whole bunch of "error: module not found:".  Please, what do?
google has not helped.

PPAC37, 2021-12-29:
Sorry I'm not familiar with Eclipse.
Modules not found, are they in the list of https://github.com/MarginallyClever/Makelangelo-software/blob/master/src/main/java/module-info.java

because it seems to be a file taken into account to generate the javadoc.
And not long ago a commit (I don't know for what reasons) reduced it.

for example, this has disappeared :
```
	exports com.marginallyclever.makelangelo;
	exports com.marginallyclever.makelangelo.select;
	exports com.marginallyclever.makelangelo.turtle;
	exports com.marginallyclever.convenience;
```

i-make-robots, 2021-12-29:
Full error message:
```
C:\Users\aggra\Documents\GitHub\makelangelo-software\src\main\java\module-info.java:5: error: module not found: org.apache.commons.io
	requires org.apache.commons.io;
	                           ^
C:\Users\aggra\Documents\GitHub\makelangelo-software\src\main\java\module-info.java:6: error: module not found: org.json
	requires org.json;
	            ^
C:\Users\aggra\Documents\GitHub\makelangelo-software\src\main\java\module-info.java:7: error: module not found: org.jetbrains.annotations
	requires org.jetbrains.annotations;
	                      ^
C:\Users\aggra\Documents\GitHub\makelangelo-software\src\main\java\module-info.java:8: error: module not found: org.slf4j
	requires org.slf4j;
	            ^
C:\Users\aggra\Documents\GitHub\makelangelo-software\src\main\java\module-info.java:9: error: module not found: logback.core
	requires logback.core;
	                ^
C:\Users\aggra\Documents\GitHub\makelangelo-software\src\main\java\module-info.java:11: error: module not found: jrpicam
	requires transitive jrpicam;
	                    ^
C:\Users\aggra\Documents\GitHub\makelangelo-software\src\main\java\module-info.java:12: error: module not found: jogamp.fat
	requires transitive jogamp.fat;
	                          ^
C:\Users\aggra\Documents\GitHub\makelangelo-software\src\main\java\module-info.java:13: error: module not found: kabeja
	requires transitive kabeja;
	                    ^
C:\Users\aggra\Documents\GitHub\makelangelo-software\src\main\java\module-info.java:14: error: module not found: jssc
	requires transitive jssc;
	                    ^
C:\Users\aggra\Documents\GitHub\makelangelo-software\src\main\java\module-info.java:15: error: module not found: vecmath
	requires transitive vecmath;
	                    ^
C:\Users\aggra\Documents\GitHub\makelangelo-software\src\main\java\module-info.java:16: error: module not found: batik.all
	requires transitive batik.all;
	                         ^
C:\Users\aggra\Documents\GitHub\makelangelo-software\src\main\java\module-info.java:17: error: module not found: xml.apis.ext
	requires transitive xml.apis.ext;
	                            ^
```

PPAC37, 2021-12-29:
I speak without knowing but
can be that under ecplise even if it is not necessary, it is necessary to have the javadocs of the dependencies?
under netbean if I right click on the dependencies, I have a "download javadoc" to retrieve the javadocs for each dependency.

cause this seems to be all the require in the module-info.java

coliss86, 2021-12-29:
I'm using inteelij and the module-info.java is not required. 
Can you post a screenshot of eclipse requiring this file ? I think it is a warning rather a requirement as the application is building and running without it.

To generate the javadoc, you should use maven : `./mvnw javadoc:javadoc`

coliss86, 2021-12-29:
I have upgraded the maven plugin to make it work in #430

i-make-robots, 2021-12-30:
Javadoc now appears under `${project.basedir}\target\site\apidocs`

closed 2021-12-30 by i-make-robots