MarginallyClever/Makelangelo-software #419: Maven Build failing SelectGUITest with HeadlessException

i-make-robots, 2021-12-27:
**Steps to make the bug happen**
1. Eclipse > Project > Debug As > Maven Build

**What was supposed to happen**
Successful run of all tests, build project.

**Platform (please complete the following information):**
 - Version: 7.31.0
 - OS: Windows

**Log file**
[INFO] Scanning for projects...
[INFO] 
[INFO] ------------------< com.marginallyclever:Makelangelo >------------------
[INFO] Building Makelangelo 7.31.0
[INFO] --------------------------------[ jar ]---------------------------------
[INFO] 
[INFO] --- maven-enforcer-plugin:3.0.0-M3:enforce (enforce-maven) @ Makelangelo ---
[INFO] 
[INFO] --- maven-enforcer-plugin:3.0.0-M3:enforce (enforce-java) @ Makelangelo ---
[INFO] 
[INFO] --- maven-resources-plugin:3.2.0:resources (default-resources) @ Makelangelo ---
[INFO] Using 'UTF-8' encoding to copy filtered resources.
[INFO] Using 'UTF-8' encoding to copy filtered properties files.
[INFO] Copying 1 resource
[INFO] Copying 19 resources
[INFO] The encoding used to copy filtered properties files have not been set. This means that the same encoding will be used to copy filtered properties files as when copying other filtered resources. This might not be what you want! Run your build with --debug to see which files might be affected. Read more at https://maven.apache.org/plugins/maven-resources-plugin/examples/filtering-properties-files.html
[INFO] 
[INFO] --- maven-compiler-plugin:3.8.1:compile (default-compile) @ Makelangelo ---
[INFO] Nothing to compile - all classes are up to date
[INFO] 
[INFO] --- maven-resources-plugin:3.2.0:testResources (default-testResources) @ Makelangelo ---
[INFO] Using 'UTF-8' encoding to copy filtered resources.
[INFO] Using 'UTF-8' encoding to copy filtered properties files.
[INFO] Copying 24 resources
[INFO] 
[INFO] --- maven-compiler-plugin:3.8.1:testCompile (default-testCompile) @ Makelangelo ---
[INFO] Nothing to compile - all classes are up to date
[INFO] 
[INFO] --- maven-surefire-plugin:3.0.0-M5:test (default-test) @ Makelangelo ---
[INFO] 
[INFO] -------------------------------------------------------
[INFO]  T E S T S
[INFO] -------------------------------------------------------
[INFO] Running com.marginallyclever.communications.SerialTransportLayerTest
No serial connections found.  Test inconclusive.
[INFO] Tests run: 1, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.193 s - in com.marginallyclever.communications.SerialTransportLayerTest
[INFO] Running com.marginallyclever.makelangelo.MakelangeloTest
version 7.31.0
[INFO] Tests run: 1, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.005 s - in com.marginallyclever.makelangelo.MakelangeloTest
[INFO] Running com.marginallyclever.makelangelo.PreferencesHelperTest
[INFO] Tests run: 1, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.065 s - in com.marginallyclever.makelangelo.PreferencesHelperTest
[INFO] Running com.marginallyclever.makelangelo.SelectDoubleTest
[INFO] Tests run: 1, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 1.044 s - in com.marginallyclever.makelangelo.SelectDoubleTest
[INFO] Running com.marginallyclever.makelangelo.SelectGUITest
[ERROR] Tests run: 1, Failures: 0, Errors: 1, Skipped: 0, Time elapsed: 0.003 s <<< FAILURE! - in com.marginallyclever.makelangelo.SelectGUITest
[ERROR] com.marginallyclever.makelangelo.SelectGUITest  Time elapsed: 0.003 s  <<< ERROR!
java.awt.HeadlessException
	at com.marginallyclever.makelangelo/com.marginallyclever.makelangelo.SelectGUITest.beforeAll(SelectGUITest.java:20)

[INFO] Running com.marginallyclever.makelangelo.TranslatorTests
[Deutsch, English, Pig Latin, Chinese, Nederlands]
[Deutsch, English, Pig Latin, Chinese, Nederlands]
current language=English
[INFO] Tests run: 3, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.106 s - in com.marginallyclever.makelangelo.TranslatorTests
[INFO] Running com.marginallyclever.makelangeloRobot.MakelangeloRobotPanelTest
[INFO] Tests run: 1, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.229 s - in com.marginallyclever.makelangeloRobot.MakelangeloRobotPanelTest
[INFO] Running com.marginallyclever.makelangeloRobot.PaperTest
[INFO] Tests run: 1, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.007 s - in com.marginallyclever.makelangeloRobot.PaperTest
[INFO] 
[INFO] Results:
[INFO] 
[ERROR] Errors: 
[ERROR]   SelectGUITest.beforeAll:20 » Headless
[INFO] 
[ERROR] Tests run: 10, Failures: 0, Errors: 1, Skipped: 0
[INFO] 
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time:  4.116 s
[INFO] Finished at: 2021-12-27T12:32:18-08:00
[INFO] ------------------------------------------------------------------------
[ERROR] Failed to execute goal org.apache.maven.plugins:maven-surefire-plugin:3.0.0-M5:test (default-test) on project Makelangelo: There are test failures.
[ERROR] 
[ERROR] Please refer to C:\Users\aggra\Documents\GitHub\makelangelo-software\target\surefire-reports for the individual test results.
[ERROR] Please refer to dump files (if any exist) [date].dump, [date]-jvmRun[N].dump and [date].dumpstream.
[ERROR] -> [Help 1]
[ERROR] 
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoFailureException

**Additional context**
Setting CI=true in Debug Configurations does not fix the issue.

PPAC37, 2021-12-27:
there is no need of the JFrame ( in testColor use directly panel )

or else you get java.awt.HeadlessException on new JFrame(...) in beforeAll
	at java.desktop/java.awt.GraphicsEnvironment.checkHeadless(GraphicsEnvironment.java:166)

i-make-robots, 2021-12-28:
Nicely done.

closed 2021-12-28 by i-make-robots