MarginallyClever/Makelangelo-software #421: Change "Show travel moves" in two places ... problems.

PPAC37, 2021-12-28:
"View" -> "Show travel moves" does not work if you have done before "File" -> "Preferences", "Graphics" tab, check or uncheck "Show travel moves".

**Steps to make the bug happen**

menu bar: "Generate art" -> (something that has displacements eg:) "Papercraft Box", closes the "Papercraft Box" window.

Identify whether or not we can see the movements (green lines)

Test menu bar "View" -> "Show travel moves" (try to checked or unchecked and see if the movements appear or disappear.)

Then go to "File" Preferences -> "Preferences" window, "Graphics" tab, check or uncheck "Show travel moves".

And re-test the menu bar "View" -> "Show travel moves" (here at home it does not work anymore.)

You must then reopen "File" -> "Preferences" if you clicked on the menu bar "View" -> "Show travel moves" for the change to take effect.

It seems to me that since it is the same concept, the two CheckBox (the one in the menu a JMenuItemCheckBox and the one in the preferences window JCheckBox) should be built on the same instance of an Action or AbstractAction.

**What was supposed to happen**
*A clear and concise description of what you expected to happen.*
whatever Check you use, the other is up to date and working.

*A clear and concise description of what happens instead.*
A kind of lock because the two are not in the same state

i-make-robots, 2021-12-28:
You are correct.  it was useless burried in graphics options.  the option should not be in graphics options.  I will clean it up.

i-make-robots, 2021-12-28:
I take it back.  I see in Makelangelo L~528....
'''java
JCheckBoxMenuItem checkboxShowPenUpMoves = new JCheckBoxMenuItem(Translator.get("GFXPreferences.showPenUp"),GFXPreferences.getShowPenUp());
checkboxShowPenUpMoves.addActionListener((e) -> {
	boolean b = !GFXPreferences.getShowPenUp();
	checkboxShowPenUpMoves.setSelected(b);
	GFXPreferences.setShowPenUp(b);
});
menu.add(checkboxShowPenUpMoves);
'''
The underlying data is stored in the same place.  it's the button state on the menu that is wrong.

i-make-robots, 2021-12-28:
Fixed in e52f11df

i-make-robots, 2021-12-28:
My solution is ugly because it's nearly the same code in two places.  I didn't know where to start making Actions, that's a larger discussion.  So this will work for the time being and can be cleaned up more later.

PPAC37, 2021-12-29:
If that fixes the trick, whether it looks good or not doesn't matter to the user. It is for the coder who will have to do the maintenace that it is necessary to add a small comment of the kind: not well factorize same thing done in ... to be reviewed.

I do not yet have enough experience to offer a good software architecture.
I would say to start where we need to and we will see later to bring together in a nice class the actions when possible and if we see an advantage.

When I am done with the firmwareUpload I try to propose an implementation of an AbstractAction which will be used in the constructor of the JCheckBox and the JMenuItemCheckBox to factorize the code if it is not too delicate. and we can then discuss it.

i-make-robots, 2021-12-29:
I suspect it will be "start making Actions and figure out how to organize them later."
As long as they are well-written and testable, I'm fine.

i-make-robots, 2021-12-30:
Is this fixed for you now?

PPAC37, 2021-12-30:
Yes it's fix so it can be close Thanks

closed 2021-12-30 by PPAC37