MarginallyClever/Makelangelo-software #425: m114 parse error

coliss86, 2021-12-29:
**Steps to make the bug happen**
1. Home a robot
2. in logs the following appears:
```
2021-12-29 12:57:04,922 DEBUG c.m.m.p.p.MarlinInterface - MarlinInterface connected. 
2021-12-29 12:57:11,566 ERROR c.m.m.p.p.MarlinPlotterInterface - M114 error: X:0.00 Y:-186.00 Z:200.00  
java.lang.NumberFormatException: For input string: "0.00 Y"
	at java.base/jdk.internal.math.FloatingDecimal.readJavaFormatString(FloatingDecimal.java:2054)
	at java.base/jdk.internal.math.FloatingDecimal.parseDouble(FloatingDecimal.java:110)
	at java.base/java.lang.Double.parseDouble(Double.java:651)
	at com.marginallyclever.makelangelo.plotter.plotterControls.MarlinPlotterInterface.onHearM114(MarlinPlotterInterface.java:116)
	at com.marginallyclever.makelangelo.plotter.plotterControls.MarlinPlotterInterface.onDataReceived(MarlinPlotterInterface.java:97)
	at com.marginallyclever.makelangelo.plotter.plotterControls.MarlinInterface.lambda$setupNetworkListener$2(MarlinInterface.java:108)
	at com.marginallyclever.communications.NetworkSession.notifyListeners(NetworkSession.java:43)
	at com.marginallyclever.communications.NetworkSession.notifyDataReceived(NetworkSession.java:51)
	at com.marginallyclever.communications.serial.SerialConnection.serialEvent(SerialConnection.java:87)
	at jssc.SerialPort$LinuxEventThread.run(SerialPort.java:1381)
2021-12-29 12:57:16,040 DEBUG c.m.m.p.p.ChooseConnection - ChooseConnection closed. 
2021-12-29 12:57:16,048 DEBUG c.m.m.p.p.MarlinInterface - MarlinInterface disconnected. 
```

**What was supposed to happen**
A clear and concise description of what you expected to happen.
A clear and concise description of what happens instead.

**Platform (please complete the following information):**
 - Version: 7.31.0
 - OS: macos
 - Robot model (if any): [e.g. Makelangelo 5]

coliss86, 2021-12-29:
I'll fix it and add unit tests

i-make-robots, 2021-12-29:
message.split("\b"); is wrong.  \s is the word boundary.  \s is "whitespace", all space, tab, \n, \r, etc.

i-make-robots, 2021-12-29:
tested with https://regex101.com/

coliss86, 2021-12-29:
yes `\b` is wrong.
I see that you have directly corrected the bug on master, why didn't you add a comment or, better, a suggestion on the PR ?

i-make-robots, 2021-12-29:
because I'm dumb.

coliss86, 2021-12-29:
ok never mind :)

commit 2437481c (2021-12-30): Merge pull request #426 from coliss86/parse

Fix #425 and add unit tests
commit 97ab28cb (2021-12-30): Fix #425 and add unit tests
closed 2021-12-30 by i-make-robots