MarginallyClever/Makelangelo-software #428: Paper Margin should work like scale (with mm/% option)

i-make-robots, 2021-12-29:
**Is your feature request related to a problem? Please describe.**
Margin with a slider works.  Could be better.

**Describe the solution you'd like**
Scale image offers % or mm choices, in both X and Y.  Margin could do the same for more fine grain control.  It should also remember each setting as a preference.

**Describe alternatives you've considered**
Leaving it alone?

**Additional context**
current paper settings
[image]

current scale settings
[image]

also note vastly different layout and margins.  consistent look and feel! :(

i-make-robots, 2021-12-29:
I thouhght "<- ->" was replaced with "landscape"?

PPAC37, 2021-12-29:
> I thouhght "<- ->" was replaced with "landscape"?

in https://github.com/MarginallyClever/Makelangelo-software/pull/369 ?

coliss86, 2021-12-29:
Yes it is in #369

i-make-robots, 2022-02-02:
Okay... trying to implement this is not easy.  the layout of the other buttons (the whole Select* system) is fighting the layout that it should have.

closed 2022-04-11 by github-actions[bot]