MarginallyClever/Makelangelo-software #431: Sometimes a StackOverflowError under Paper Settings.

PPAC37, 2021-12-30:
In the "paper Settings" window, I have a bug that seems to happen every now and then but I can't seem to find a way to trigger it more than once in ~ 20 tries.

**Steps to make the bug happen**
play "very actively" (repeat several times) in the "PaperSettings":
* change of "Paper Size" (ex A4)
* then "landscape" (check or uncheck)
* then "Paper color" (select another color)
* then re change the "Paper Size" (ex A2)

**What was supposed to happen**
...
but 
from time to time i get an infinite loop of PropertyChangeEvent / firePropertyChange which gives a StackOverflowError

**Platform (please complete the following information):**
 - Version: dev https://github.com/MarginallyClever/Makelangelo-software/commit/eedc8c2e52dab426da0c3dac72e52dd61a26fda6
 - OS: Ubuntu 18.04 
 
**Log file**
the beginning of the exception because the stack is rather large ... but the repeting patern is not exactly the same ( lines change ...)

```
Exception in thread "AWT-EventQueue-0" java.lang.StackOverflowError
	at java.base/java.lang.Thread.setPriority(Thread.java:1144)
	at java.base/java.lang.Thread.<init>(Thread.java:446)
	at java.base/java.lang.Thread.<init>(Thread.java:708)
	at java.base/java.lang.Thread.<init>(Thread.java:477)
	at java.base/java.util.TimerThread.<init>(Timer.java:510)
	at java.base/java.util.Timer.<init>(Timer.java:103)
	at java.base/java.util.Timer.<init>(Timer.java:170)
	at com.marginallyclever.makelangelo.select.SelectDouble$1.validate(SelectDouble.java:77)
	at com.marginallyclever.makelangelo.select.SelectDouble$1.insertUpdate(SelectDouble.java:59)
	at java.desktop/javax.swing.text.AbstractDocument.fireInsertUpdate(AbstractDocument.java:226)
	at java.desktop/javax.swing.text.AbstractDocument.handleInsertString(AbstractDocument.java:780)
	at java.desktop/javax.swing.text.AbstractDocument.insertString(AbstractDocument.java:739)
	at java.desktop/javax.swing.text.PlainDocument.insertString(PlainDocument.java:131)
	at java.desktop/javax.swing.text.AbstractDocument.replace(AbstractDocument.java:698)
	at java.desktop/javax.swing.text.JTextComponent.setText(JTextComponent.java:1729)
	at com.marginallyclever.makelangelo.select.SelectDouble.setValue(SelectDouble.java:123)

	at com.marginallyclever.makelangelo.paper.PaperSettings.onLandscapeChange(PaperSettings.java:126)
	at com.marginallyclever.makelangelo.paper.PaperSettings.lambda$new$6(PaperSettings.java:97)
	at com.marginallyclever.makelangelo.select.Select.firePropertyChange(Select.java:38)
	at com.marginallyclever.makelangelo.select.SelectBoolean.lambda$new$0(SelectBoolean.java:33)
	at java.desktop/javax.swing.AbstractButton.fireItemStateChanged(AbstractButton.java:2000)
	at java.desktop/javax.swing.AbstractButton$Handler.itemStateChanged(AbstractButton.java:2320)
	at java.desktop/javax.swing.DefaultButtonModel.fireItemStateChanged(DefaultButtonModel.java:458)
	at java.desktop/javax.swing.JToggleButton$ToggleButtonModel.setSelected(JToggleButton.java:374)
	at java.desktop/javax.swing.AbstractButton.setSelected(AbstractButton.java:346)
	at com.marginallyclever.makelangelo.select.SelectBoolean.setSelected(SelectBoolean.java:46)
	at com.marginallyclever.makelangelo.paper.PaperSettings.onPaperDimensionsChange(PaperSettings.java:110)
	at com.marginallyclever.makelangelo.paper.PaperSettings.lambda$new$2(PaperSettings.java:93)
	at com.marginallyclever.makelangelo.select.Select.firePropertyChange(Select.java:38)
	at com.marginallyclever.makelangelo.select.SelectDouble.setValue(SelectDouble.java:124)

	at com.marginallyclever.makelangelo.paper.PaperSettings.onLandscapeChange(PaperSettings.java:127)
	at com.marginallyclever.makelangelo.paper.PaperSettings.lambda$new$6(PaperSettings.java:97)
	at com.marginallyclever.makelangelo.select.Select.firePropertyChange(Select.java:38)
	at com.marginallyclever.makelangelo.select.SelectBoolean.lambda$new$0(SelectBoolean.java:33)
	at java.desktop/javax.swing.AbstractButton.fireItemStateChanged(AbstractButton.java:2000)
	at java.desktop/javax.swing.AbstractButton$Handler.itemStateChanged(AbstractButton.java:2320)
	at java.desktop/javax.swing.DefaultButtonModel.fireItemStateChanged(DefaultButtonModel.java:458)
	at java.desktop/javax.swing.JToggleButton$ToggleButtonModel.setSelected(JToggleButton.java:374)
	at java.desktop/javax.swing.AbstractButton.setSelected(AbstractButton.java:346)
	at com.marginallyclever.makelangelo.select.SelectBoolean.setSelected(SelectBoolean.java:46)
	at com.marginallyclever.makelangelo.paper.PaperSettings.onPaperDimensionsChange(PaperSettings.java:115)
	at com.marginallyclever.makelangelo.paper.PaperSettings.lambda$new$2(PaperSettings.java:93)
	at com.marginallyclever.makelangelo.select.Select.firePropertyChange(Select.java:38)
	at com.marginallyclever.makelangelo.select.SelectDouble.setValue(SelectDouble.java:124)

	at com.marginallyclever.makelangelo.paper.PaperSettings.onLandscapeChange(PaperSettings.java:127)
	at com.marginallyclever.makelangelo.paper.PaperSettings.lambda$new$6(PaperSettings.java:97)
	at com.marginallyclever.makelangelo.select.Select.firePropertyChange(Select.java:38)
	at com.marginallyclever.makelangelo.select.SelectBoolean.lambda$new$0(SelectBoolean.java:33)
	at java.desktop/javax.swing.AbstractButton.fireItemStateChanged(AbstractButton.java:2000)
	at java.desktop/javax.swing.AbstractButton$Handler.itemStateChanged(AbstractButton.java:2320)
	at java.desktop/javax.swing.DefaultButtonModel.fireItemStateChanged(DefaultButtonModel.java:458)
	at java.desktop/javax.swing.JToggleButton$ToggleButtonModel.setSelected(JToggleButton.java:374)
	at java.desktop/javax.swing.AbstractButton.setSelected(AbstractButton.java:346)
	at com.marginallyclever.makelangelo.select.SelectBoolean.setSelected(SelectBoolean.java:46)
	at com.marginallyclever.makelangelo.paper.PaperSettings.onPaperDimensionsChange(PaperSettings.java:110)
	at com.marginallyclever.makelangelo.paper.PaperSettings.lambda$new$2(PaperSettings.java:93)
	at com.marginallyclever.makelangelo.select.Select.firePropertyChange(Select.java:38)
	at com.marginallyclever.makelangelo.select.SelectDouble.setValue(SelectDouble.java:124)

	at com.marginallyclever.makelangelo.paper.PaperSettings.onLandscapeChange(PaperSettings.java:127)
	at com.marginallyclever.makelangelo.paper.PaperSettings.lambda$new$6(PaperSettings.java:97)
	at com.marginallyclever.makelangelo.select.Select.firePropertyChange(Select.java:38)
	at com.marginallyclever.makelangelo.select.SelectBoolean.lambda$new$0(SelectBoolean.java:33)
	at java.desktop/javax.swing.AbstractButton.fireItemStateChanged(AbstractButton.java:2000)
...
```

Sorry if my last "PR" "paper_pref" introduced this behavior. (I encountered this in my dev and I thought I had isolated the reason but now I realize that no I missed the thing. And so I don't know if it was present before or not (I have to test, check on the release version.))

PPAC37, 2021-12-30:
Impossible to have this with the "Makelangelo-7.28.2-with-dependencies.jar"
Because, the landscape has another concern (does not change state).

But problem present on the makelangelo-nightly-20211230-243748.jar

i-make-robots, 2021-12-30:
Some Select* items were firing propertyChangeEvents, which could - in the right conditions - cause a loop.  
Setting a value in a Select should not fire an event automatically.  
I have removed the firePropertyChangeEvents from all Select in c391867f
In the process I noticed that clicking the landscape button did not immediately update the preview.  fixed that, too.

PPAC37, 2021-12-30:
Perfect. You fixed the problem.
After many tests (more than 20x the cycle previously described) nothing to report.
Well done!
For me this issue can be closed.

closed 2021-12-30 by PPAC37