MarginallyClever/Makelangelo-software #434: Improve log for diagnostic

coliss86, 2022-01-05:
**Is your feature request related to a problem? Please describe.**
Recent discussions on discord show that some information are missing for user and developers to help each other.

**Describe the solution you'd like**
- I think of adding the version of makelangelo when the version of marlin is detected in the output of the robot will help.
- makelangelo should also detect when the robot says `echo:Home XY First` and display a dialog box to the user.
- same thing when makelangelo logs `MarlinInterface did not find 0`

**Describe alternatives you've considered**
None

**Additional context**
None

i-make-robots, 2022-01-05:
Master branch I turned on logging of all marlin to app conversation.  that will capture the marlin version in the first message.

I thought Makelangelo version was listed above the system settings at the start of the log file, but you're right!  I don't find it.

coliss86, 2022-01-05:
I was saying about adding the version of makelangelo in the debug output of the robot:
[image]

it is already in the startup log of the application: 
[image]

i-make-robots, 2022-01-05:
Yes, yes.  So everything here is already implemented.  Let's leave this open for a bit in case new ideas come up.

i-make-robots, 2022-01-25:
Nothing?  Cool.

closed 2022-01-25 by i-make-robots