MarginallyClever/Makelangelo-software #446: "robot controls > advanced controls > home" does not unlock draw control buttons

i-make-robots, 2022-01-12:
**Steps to make the bug happen**
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**What was supposed to happen**
A clear and concise description of what you expected to happen.
A clear and concise description of what happens instead.

**Platform (please complete the following information):**
 - 7.31.0 bugfix
 - Windows
 - M5

**Log file**
Makelangelo software should generate a log.txt file.  Please paste or link it here.

**Additional context**
Add any other context about the problem here.
If applicable, add screenshots to help explain your problem.

coliss86, 2022-01-13:
Is the robot connected when you click on the home button ?

i-make-robots, 2022-01-13:
Of course.

PR #447 (2022-01-17): Fix #446: unlock draw control buttons when find home button is clicked
Fix #446:
> "robot controls > advanced controls > home" does not unlock draw control buttons
commit 67471923 (2022-01-17): Merge pull request #447 from coliss86/fix

Fix #446: unlock draw control buttons when find home button is clicked
commit 7b18af9b (2022-01-17): Fix #446: unlock draw control buttons when find home button is clicked
commit 2e1d2128 (2022-01-17): Merge pull request #449 from coliss86/resize

Fix #446: the robot control window is not resizable anymore
closed 2022-01-17 by i-make-robots