MarginallyClever/Makelangelo-software #448: ERROR in ch.qos.logback.core.joran.action.NestedComplexPropertyIA

i-make-robots, 2022-01-13:
I start the application from Eclipse Debug, and get the following output in the serial window:
```
09:57:56,028 |-INFO in ch.qos.logback.classic.LoggerContext[default] - Could NOT find resource [logback-test.xml]
09:57:56,028 |-INFO in ch.qos.logback.classic.LoggerContext[default] - Found resource [logback.xml] at [file:/C:/Users/aggra/Documents/GitHub/makelangelo-software/target/classes/logback.xml]
09:57:56,082 |-INFO in ch.qos.logback.classic.joran.action.ConfigurationAction - debug attribute not set
09:57:56,205 |-INFO in ch.qos.logback.core.joran.action.AppenderAction - About to instantiate appender of type [ch.qos.logback.core.rolling.RollingFileAppender]
09:57:56,209 |-INFO in ch.qos.logback.core.joran.action.AppenderAction - Naming appender as [FILE]
09:57:56,221 |-INFO in ch.qos.logback.core.rolling.FixedWindowRollingPolicy@1cbb87f3 - No compression will be used
09:57:56,223 |-ERROR in ch.qos.logback.core.joran.action.NestedComplexPropertyIA - Could not create component [triggeringPolicy] of type [com.marginallyclever.convenience.log.RollOncePerSessionTriggeringPolicy] java.lang.IllegalAccessException: class ch.qos.logback.core.joran.action.NestedComplexPropertyIA (in module logback.core) cannot access class com.marginallyclever.convenience.log.RollOncePerSessionTriggeringPolicy (in module com.marginallyclever.makelangelo) because module com.marginallyclever.makelangelo does not export com.marginallyclever.convenience.log to module logback.core
	at java.lang.IllegalAccessException: class ch.qos.logback.core.joran.action.NestedComplexPropertyIA (in module logback.core) cannot access class com.marginallyclever.convenience.log.RollOncePerSessionTriggeringPolicy (in module com.marginallyclever.makelangelo) because module com.marginallyclever.makelangelo does not export com.marginallyclever.convenience.log to module logback.core
	at 	at java.base/jdk.internal.reflect.Reflection.newIllegalAccessException(Reflection.java:385)
	at 	at java.base/java.lang.reflect.AccessibleObject.checkAccess(AccessibleObject.java:693)
	at 	at java.base/java.lang.reflect.Constructor.newInstanceWithCaller(Constructor.java:490)
	at 	at java.base/java.lang.reflect.ReflectAccess.newInstance(ReflectAccess.java:128)
	at 	at java.base/jdk.internal.reflect.ReflectionFactory.newInstance(ReflectionFactory.java:350)
	at 	at java.base/java.lang.Class.newInstance(Class.java:645)
	at 	at logback.core@1.2.10/ch.qos.logback.core.joran.action.NestedComplexPropertyIA.begin(NestedComplexPropertyIA.java:121)
	at 	at logback.core@1.2.10/ch.qos.logback.core.joran.spi.Interpreter.callBeginAction(Interpreter.java:269)
	at 	at logback.core@1.2.10/ch.qos.logback.core.joran.spi.Interpreter.startElement(Interpreter.java:145)
	at 	at logback.core@1.2.10/ch.qos.logback.core.joran.spi.Interpreter.startElement(Interpreter.java:128)
	at 	at logback.core@1.2.10/ch.qos.logback.core.joran.spi.EventPlayer.play(EventPlayer.java:50)
	at 	at logback.core@1.2.10/ch.qos.logback.core.joran.GenericConfigurator.doConfigure(GenericConfigurator.java:165)
	at 	at logback.core@1.2.10/ch.qos.logback.core.joran.GenericConfigurator.doConfigure(GenericConfigurator.java:152)
	at 	at logback.core@1.2.10/ch.qos.logback.core.joran.GenericConfigurator.doConfigure(GenericConfigurator.java:110)
	at 	at logback.core@1.2.10/ch.qos.logback.core.joran.GenericConfigurator.doConfigure(GenericConfigurator.java:53)
	at 	at ch.qos.logback.classic.util.ContextInitializer.configureByResource(ContextInitializer.java:64)
	at 	at ch.qos.logback.classic.util.ContextInitializer.autoConfig(ContextInitializer.java:134)
	at 	at org.slf4j.impl.StaticLoggerBinder.init(StaticLoggerBinder.java:84)
	at 	at org.slf4j.impl.StaticLoggerBinder.<clinit>(StaticLoggerBinder.java:55)
	at 	at org.slf4j@1.7.32/org.slf4j.LoggerFactory.bind(LoggerFactory.java:150)
	at 	at org.slf4j@1.7.32/org.slf4j.LoggerFactory.performInitialization(LoggerFactory.java:124)
	at 	at org.slf4j@1.7.32/org.slf4j.LoggerFactory.getILoggerFactory(LoggerFactory.java:417)
	at 	at org.slf4j@1.7.32/org.slf4j.LoggerFactory.getLogger(LoggerFactory.java:362)
	at 	at org.slf4j@1.7.32/org.slf4j.LoggerFactory.getLogger(LoggerFactory.java:388)
	at 	at com.marginallyclever.makelangelo/com.marginallyclever.convenience.log.Log.start(Log.java:28)
	at 	at com.marginallyclever.makelangelo/com.marginallyclever.makelangelo.Makelangelo.main(Makelangelo.java:883)
09:57:56,223 |-INFO in ch.qos.logback.core.joran.action.NestedComplexPropertyIA - Assuming default type [ch.qos.logback.classic.encoder.PatternLayoutEncoder] for [encoder] property
09:57:56,263 |-WARN in ch.qos.logback.core.rolling.RollingFileAppender[FILE] - No TriggeringPolicy was set for the RollingFileAppender named FILE
09:57:56,263 |-WARN in ch.qos.logback.core.rolling.RollingFileAppender[FILE] - For more information, please visit http://logback.qos.ch/codes.html#rfa_no_tp
09:57:56,264 |-INFO in ch.qos.logback.classic.joran.action.LoggerAction - Setting level of logger [com.marginallyclever] to DEBUG
09:57:56,264 |-INFO in ch.qos.logback.classic.joran.action.RootLoggerAction - Setting level of ROOT logger to INFO
09:57:56,264 |-INFO in ch.qos.logback.core.joran.action.AppenderRefAction - Attaching appender named [FILE] to Logger[ROOT]
09:57:56,264 |-INFO in ch.qos.logback.classic.joran.action.ConfigurationAction - End of configuration.
09:57:56,264 |-INFO in ch.qos.logback.classic.joran.JoranConfigurator@22e357dc - Registering current configuration as safe fallback point```

PPAC37, 2022-01-17:
A debug running under NetBeans seems to work... sorry, I have no leads

coliss86, 2022-01-17:
On my setup, I have to remove the file `module-info.java` to be able to start the application and thus no error is reported when running in debug mode.
I think this file is not necessary for runtime.

closed 2022-01-17 by i-make-robots