MarginallyClever/Makelangelo-software #458: can't run the app with module-info.java

coliss86, 2022-01-17:
**Steps to make the bug happen**
If I run the app with Intellij, it fails with the following error: 
```
Error occurred during initialization of boot layer
java.lang.module.FindException: Unable to derive module descriptor for /Users/guillaume/.m2/repository/xalan/xalan/2.7.2/xalan-2.7.2.jar
Caused by: java.lang.module.InvalidModuleDescriptorException: Provider class org.apache.bsf.BSFManager not in module

Process finished with exit code 1
```

**What was supposed to happen**
The app should starts

**Platform (please complete the following information):**
 - Version: 7.31.0
 - OS: OS X, 11.6.1
 - Robot model (if any): [e.g. Makelangelo 5]

**Additional context**
If I run it with the jar, no error is reported, like : 
`java -jar Makelangelo-7.31.0-with-dependencies.jar`

@i-make-robots @PPAC37 what is your configuration ? OS, java ?

i-make-robots, 2022-01-17:
Windows 10, Eclipse 2021-12 (4.22.0), jdk-15.0.2

PPAC37, 2022-01-18:
Ubuntu 18.04, NetBeans IDE 12.6, java 17 2021-09-14 LTS.

It seems to me that the notion of module (therefore module-info.java) is normally for JAVA 9 (but it's not yet clear to me because I haven't played with java 9 yet).

The only thing I'm sure of,
is that to have a javadoc not too empty, we need module-info.java ... (but not really important. Facilitating development remains a priority.)

Otherwise NetBeans 12.6 GUI notifies me of the com.marginallyclever.makelangelo module in error but that doesn't seem to cause it any more problems...
[image]

The tooltips content on the error (!) :
```
module com.marginallyclever.makelangelo reads package me.saharnooby.qoi from both qoi.java and qoi.java.awt
```
and it warns me almost all transitive requires (except for jssc)
The tooltips content on the warms /!\ :
```
requires transitive directive for an automatic module
```

So I'm also confused. 
Sorry I don't know what to look for/correct.

coliss86, 2022-01-18:
Thx both for your answer

coliss86, 2022-01-20:
I can't get it working with file `module-info.java` with IntelliJ but it may be related to my setup. 
Closing it as I can't find a solution but it doesn't impact real users.

closed 2022-01-20 by coliss86