MarginallyClever/Makelangelo-software #460: MenuBar : File -> Save = Error (but file created ans seems ok ...)

PPAC37, 2022-01-19:
A throw out of context ? 

https://github.com/MarginallyClever/Makelangelo-software/blob/33cb237f34463936f6ab07ce84339bff82bf416b/src/main/java/com/marginallyclever/makelangelo/makeArt/io/vector/TurtleFactory.java#L75

i-make-robots, 2022-01-20:
What format were you saving as?  Please show steps to recreate the error.

PPAC37, 2022-01-20:
My apologies (bad presentation of the bug)
Makelangelo 7.31.0 (master)
(Possibly do a "Generate art" but does not change anything)
Then, menu bar "File" -> "Save"
[image]

On save file dialog : 

* filename ex "test_d"
* Files of Type: "DXF R12" or "Scalable Vector Graphics 1.1" ( same result)
* click "Save"

[image]

Error 
TurtulFactory could not save '...'.

But if you check the destination, the file was created and looks correct...

i-make-robots, 2022-01-20:
I confirm this is replicable.

i-make-robots, 2022-01-20:
fixed on both branches.

closed 2022-01-20 by i-make-robots