MarginallyClever/Makelangelo-software #464: Remember last path from load and save

i-make-robots, 2022-01-20:
**Is your feature request related to a problem? Please describe.**
When loading a file remember the path of the file for the next time a file is loaded.
When saving a file do the same.  Each load and save should have a separate memory (load path might not be the same as save path).

**Describe the solution you'd like**
Preferences should retain the path.  

**Describe alternatives you've considered**
None.

**Additional context**
If the path no longer exists when the load/save dialog is created, default to the current working directory.

i-make-robots, 2022-01-20:
If someone else can confirm this works I can then close the issue.

closed 2022-01-20 by i-make-robots

PPAC37, 2022-01-21:
It's done in far fewer lines than I had envisioned. Well played!

( I would have implemented this directly in the SaveDialog and LoadSavePanel classes ... ( choice of architecture, but as it's done in very few lines your solution seems much cleaner. ) )

the solution you implemented memorizes even in case we "Cancel" the dialog. (But not really a problem because rarely the case and debatable whether or not to memorize the path when you do "cancel".)

Otherwise "user.dir" under linux is not "user.home" but as in the JFileChoser there is a "Home" icon no problem.

there, so it's only done on the bug-fix branch ... ( I guess it will be "merge" to the master branch later? )

commit 56300f8d (2022-01-20): use getLastPath and setLastPath to fix #464