MarginallyClever/Makelangelo-software #472: Can't click on range slider

coliss86, 2022-01-23:
**Steps to make the bug happen**
I can't select the range slider because it's to narrow of the border of the window and thus the cursor transforms itself to resize the window.
Here is a video:

https://user-images.githubusercontent.com/23615562/150691497-fcf43951-45ce-466b-9dc0-82971a7fde31.mov

**What was supposed to happen**
I can drag the blue or red button

**Platform (please complete the following information):**
 - Version: 7.31.0 master
 - OS: osx

i-make-robots, 2022-01-23:
I think the best solution is to add a JLabel on both ends.  when the slider is moved, the JLabel displays the number equivalent of the slider.  so if there are 100 commands and I love the left slider to the middle then the left JLabel would say "49".  This way users can tell exactly what line number they have selected and that might be useful for splitting files or something.

i-make-robots, 2022-01-25:
[image]

closed 2022-01-25 by i-make-robots