MarginallyClever/Makelangelo-software #474: Add options when loading a Svg

coliss86, 2022-01-23:
**Is your feature request related to a problem? Please describe.**
As with images, when loading a svg, I would like to be able to have the preview pane populated with the select menu to "fill image on the paper" and co, like here:
[image]

**Describe the solution you'd like**
I can add this dropdown menu as with image, WDYT ?

i-make-robots, 2022-01-23:
If we add Scale here you'll want to add Flip, too, and then where does it end?  It makes two ways to do the same thing.  imho bad design.

Fit is only here because bitmaps cannot be moved after being loaded.  So this opens the deeper question about how bitmaps are treated in app.  For example, if there was a system of layers (ala photoshop) or a node-based editor (see blender) then a image could be adjustable after the load, which would cause the generators to run again, etc.  Either system start to duplicate a lot of behaviour already in Inkscape or https://github.com/SonarSonic/DrawingBotV3

I don't think a close button is needed when there's already 'x' in the corner.  one option is enough.  it doesn't need translation and everyone knows what it does.

coliss86, 2022-01-23:
I don't understand: scale and flip also work for images after the LoadFilePanel, so they already are duplicated, don't they ?

This is what I have when loading a svg: an empty graphic area and two more click to set the size at the correct position
[image]

i-make-robots, 2022-01-28:
The size, scale, and position of the bitmap is related to the pen and plotter size when converting to bitmap.  The fit and flip tools here adjust the position of the bitmap before it is converted.  Once it is converted the bitmap is thrown away.  
Maybe "scale image" should say "scale bitmap" to better distinguish between the two image types.  

I would be ok with adding an automatic scale-to-fit for all vector files.

closed 2022-01-28 by i-make-robots